swri-robotics/mapviz #237: Release for ROS Jade

evenator, 2015-09-28:


evenator, 2015-09-29:
Released to OSRF build farm as version 0.1.0

cbandera, 2015-10-09:
Hi, I am having difficulties installing mapviz.

 * Trying to install from apt-get I get the error: ```E: Unable to locate package ros-jade-mapviz```
  * Running ```cat /etc/apt/sources.list.d/ros-latest.list``` prints ```deb http://packages.ros.org/ros/ubuntu trusty main```
  * I have already run a ```rosdep update```
 * When trying to install from source, and running the ```rosdep install``` command I get: `E: Unable to locate package ros-jade-marti-visualization-msgs`

Any ideas?

evenator, 2015-10-09:
The installation problems you're experiencing are because the OSRF has not yet synced Mapviz from the ros-shadow-fixed apt repository to the main ROS apt repository. In the meantime, if you want to install Mapviz, you have two options:

Option 1:
Add mapviz and its dependencies to your catkin workspace and build mapviz from source. If you're using wstool, you can add the dependencies by running two commands from the src directory of your catkin workspace:
    wstool set mapviz --git https://github.com/swri-robotics/mapviz.git -v $ROS_DISTRO-devel
    wstool set marti_common --git https://github.com/swri-robotics/marti_common.git -v $ROS_DISTRO-devel
    wstool set marti_messages --git https://github.com/swri-robotics/marti_messages.git -v indigo-devel
If you're not using ws_tool, you can do the same with git by running these two commands from the src directory of your catkin workspace:
    git clone https://github.com/swri-robotics/mapviz.git -b $ROS_DISTRO-devel
    git clone https://github.com/swri-robotics/marti_common.git -b $ROS_DISTRO-devel
    git clone https://github.com/swri-robotics/marti_messages.git -b indigo-devel
Then you should be able to build Mapviz using catkin.

Option 2:
Add the ros-shadow-fixed apt repository to your computer's apt sources. It contains prerelease ROS packages, including Mapviz and its dependencies. You can add it to your computer and install Mapviz with these three commands:
    sudo sh -c 'echo "deb http://packages.ros.org/ros-shadow-fixed/ubuntu $(lsb_release -sc) main" > /etc/apt/sources.list.d/ros-shadow.list'
    sudo apt-get update
    sudo apt-get install ros-$ROS_DISTRO-mapviz

closed 2015-10-09 by evenator