swri-robotics/mapviz #238: Update gps mapviz_plugins for Jade

evenator, 2015-09-29:
The gps plugin uses gps_common/GPSFix, which is gone in Jade. Proposed fix: add a navsat plugin to indigo and remove the gps plugin in jade.

evenator, 2015-09-29:
GPS plugin removed, but not replaced with a NavSat plugin.

jeremy-a-asi, 2015-11-16:
I'm having trouble finding a package/meta package that Uses GPSFix in Indigo. I'm currently using the Navsat (http://wiki.ros.org/nmea_navsat_driver) and nmea_comms (http://wiki.ros.org/nmea_comms) packages.

Can I run this fix with Indigo (mix Jade and Indigo)? 

Am I reading the last comment (info) by cbandera correctly that this will be closed when back ported to Indigo?

evenator, 2015-11-16:
I've been meaning to backport the NavSatFix plugin to Indigo. I'll get on that.

evenator, 2015-11-16:
I backported @cbandera's NavSat plugin to indigo. Could you check out the backport-navsat-plugin branch and test it out for me?

jeremy-a-asi, 2015-11-16:
I'm having a problem building. 

/home/user/oem_ws/src/mapviz/multires_image/src/tile_set.cpp:90: error: 'class swri_transform_util::GeoReference' has no member named 'Extension'
     m_extension = m_geo.Extension();
                         ^

I don't see the Extension method defined in georeference.h. Did I miss something?

evenator, 2015-11-16:
Make sure you have the lastest version of indigo-devel checked out in the marti_common repository.

closed 2015-11-22 by malban

evenator, 2015-12-11:
Has anyone tested the NavSat plugin in Indigo? I don't want to cut another Indigo release until it's tested.

pjreed, 2015-12-11:
I've used it in Indigo and it works for me.

commit 795a517d (2015-11-01): Fixes #238. Replaced GPSFix with NavSatFix. Orientation is ignored, as it is not part of the message anymore.