swri-robotics/mapviz #254: mapviz fails to build on armhf

tfoote, 2015-10-21:
We're spinning up arm builds armhf at the moment aarch64 coming soon. mapviz makes assumptions about the architecture which fail to build. 

http://54.183.26.131:8080/job/Ibin_arm_uThf__mapviz__ubuntu_trusty_armhf__binary/2/console
```
01:16:21.995 [ 25%] Building CXX object CMakeFiles/mapviz.dir/src/mapviz.cpp.o
01:16:22.136 /usr/lib/ccache/arm-linux-gnueabihf-g++   -DQT_CORE_LIB -DQT_GUI_LIB -DQT_NO_KEYWORDS -DQT_OPENGL_LIB -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"mapviz\" -g -O2 -fstack-protector --param=ssp-buffer-size=4 -Wformat -Werror=format-security -DNDEBUG -D_FORTIFY_SOURCE=2  -I/usr/include/opencv -isystem /usr/include/qt4 -isystem /usr/include/qt4/QtOpenGL -isystem /usr/include/qt4/QtGui -isystem /usr/include/qt4/QtCore -I/tmp/binarydeb/ros-indigo-mapviz-0.0.3/include -I/opt/ros/indigo/include -I/tmp/binarydeb/ros-indigo-mapviz-0.0.3/obj-arm-linux-gnueabihf -I/tmp/binarydeb/ros-indigo-mapviz-0.0.3 -I/tmp/binarydeb/ros-indigo-mapviz-0.0.3/src    -o CMakeFiles/mapviz.dir/src/mapviz.cpp.o -c /tmp/binarydeb/ros-indigo-mapviz-0.0.3/src/mapviz.cpp
01:18:44.180 In file included from /usr/include/qt4/QtOpenGL/QGLWidget:1:0,
01:18:44.180                  from /tmp/binarydeb/ros-indigo-mapviz-0.0.3/include/mapviz/mapviz_plugin.h:43,
01:18:44.180                  from /tmp/binarydeb/ros-indigo-mapviz-0.0.3/include/mapviz/mapviz.h:71,
01:18:44.180                  from /tmp/binarydeb/ros-indigo-mapviz-0.0.3/src/mapviz.cpp:30:
01:18:44.180 /usr/include/qt4/QtOpenGL/qgl.h:85:17: error: conflicting declaration ‘typedef GLfloat GLdouble’
01:18:44.180  typedef GLfloat GLdouble;
01:18:44.180                  ^
01:18:44.180 In file included from /tmp/binarydeb/ros-indigo-mapviz-0.0.3/include/mapviz/mapviz.h:40:0,
01:18:44.180                  from /tmp/binarydeb/ros-indigo-mapviz-0.0.3/src/mapviz.cpp:30:
01:18:44.180 /usr/include/GL/glew.h:283:16: error: ‘GLdouble’ has a previous declaration as ‘typedef double GLdouble’
01:18:44.180  typedef double GLdouble;
```

evenator, 2015-10-23:
I propose we close this as a wontfix unless someone comes forward wanting to run mapviz on an ARM. Thoughts?

malban, 2016-01-08:
I would like to fix this, but unfortunately this appears to be the result of an incompatibility between the QT OpenGL bindings and the OpenGL library on the ARM version of Ubuntu.

Fixing it will probably require replacing the use of the QOpenGLWidget canvas with something else so it doesn't have to depend on the broken QT OpenGL bindings.

pjreed, 2017-09-29:
I spent a while digging into this -- I've actually got a branch where I rewrote MapCavas to use QOpenGLWidget rather than QGLWidget on Kinetic (https://github.com/pjreed/mapviz/tree/kinetic-opengl-update) -- and I think it'd make sense to mark this as `wontfix`.

Basically, libQt5OpenGL.so on armhf is linked against libGLESv2.so rather than libGL.so, so it only has the GL ES 2.0 API available to it.  The ES API does not include any of the immediate mode rendering commands (glPushMatrix, glPopMatrix, glOrtho, etc.) that we use in all of the Mapviz plugins that do GL drawing.

That means that in order to make Mapviz build on armhf, we need to do one of two things:
- Rewrite every plugin to use retained mode for drawing, which is both a pain and would prevent Mapviz from working properly over a forwarded X11 display
- Directly link in libGL.so, strip out references to QGLWidget, and figure out how to do GL drawing directly on a surface that isn't managed by Qt, which would probably have the side effect of being incompatible with the QPainter drawing that some plugins do

Both of those have significant drawbacks and are a lot of work, and there hasn't really been any demand for mapviz on armhf.

danthony06, 2021-06-11:
I think this is the same as reported here: https://github.com/swri-robotics/mapviz/issues/660. I'm going to close this one out to remove the duplicate issue. I'm not sure that we'll support Arm builds, because typically those are running as a headless embedded device on the robots, and aren't usually where users are running Mapviz from.

closed 2021-06-11 by danthony06