swri-robotics/mapviz #256: Missing documentation on reference frames

cbandera, 2015-10-25:
Hi,

I am having difficulties to understand how different frames are handled. 

So I have a GPS Position on the one hand, which gets displayed without issues.
Also, I am trying to visualize a path msg. The points with in the path message use GPS coordinates as well, but I just can't get them to visualize correctly.

Judging from the code, the GPS position uses the `local_xy_util_` which converts the GPS coordinates into a local reference frame. How can I get my path to get converted as well? Or do I need to publish some kind of tf transform? I have tried setting the `frame_id` to origin, wgs84 or far_field but still it doesn't get displayed correctly. How is the path message supposed to be used and how would I project data like laser scans into the map?

Thx,
cbandera

malban, 2015-10-25:
cbandera,

Thanks for your interest and contributions.  I agree we need to document the reference frames better.  I'm not sure what exactly is preventing your path data to not display, but I can describe how it should work.  What is the coordinate system of the data you are putting in the path message? lat/lon or local frame?

For the case where all the data is referenced with respect to tf frames, mapviz should behave similarly to rviz.  In the standard ROS system the core frames are map->odom->base_link.  Some of our projects have used a different convention far_field->near_field->vehicle, but they mean the same thing and the ROS conventions are preferable all things equal for interoperability.

When dealing with lat/lon or other geo-referenced data we have the issue that the tf framework only supports rigid translation and rotation.  The solution we use is to anchor the origin of the 'map' frame to a geographic coordinate and define a projection from the WGS84 ellipsoid to the plane of the 'map' frame centered on the anchor point.  This provides the transform to go between the tf tree and lat/lon and back.  The local coordinate system is referred to as LocalXY and the swri_transform_util/local_xy_util.h header contains helper functions and classes for using it.

To simplify our code we also created a wrapper around the tf listener and tf transform classes to transparently handle conversions between normal tf frames and special non-tf frames like '/wgs84'.  This is what the path mapviz plug-in uses to transform data, so if the path message contains lat / lon data and has the '/wgs84' frame it will convert it to the fixed frame of mapviz.

Conceptually the transform tree looks like wgs84->map->odom->base_link, where the wgs84->map transform uses LocalXY instead of tf.

In order to perform the LocalXY transform, the LocalXY origin, or anchor point must be known.  This is published by the transform_util initialize_origin.py node, which can take a fixed coordinate or subscribe to GPS and use the first message received.

Possible issues:
 - No origin is being published
 - The wgs84 frame_id string doesn't match.  transform_util/frames.h provides a constant that can be used.
 - The map frame has not been published by anything and so is unknown to tf and the transform_util wrapper.

In most cases I would also recommend using an actual tf frame like 'map' as the fixed frame for mapviz.  In that case all data referenced in normal tf frames can be displayed with out requiring a global origin.

Marc Alban
Southwest Research Institute

malban, 2015-10-25:
You should also make sure that the local origin is relatively close to the data you are dealing with since eventually the flat-earth projection will break down.

Georacer, 2015-11-14:
Hello,

Despite reading this ticket, I still can't make tile maps work. I get the message "No transform between /wgs84 and map".

In my code, I have two publishers: one from map to local_origin, and one from local_origin to fcu.

I know the coordinates of the map frame, hence I have added this code to my launch file:
```
    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
        <param name="local_xy_frame" value="/wgs84"/>
        <param name="local_xy_origin" value="crops"/>
        <rosparam param="local_xy_origins">
            [{ name: crops,
             latitude: 37.9879772,
             longitude: 23.9078602,
             altitude: 129.69,
             heading: 0.0}]
        </rosparam>
    </node>

    <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /wgs84 /map 100"  />
```
However, the error message persists and the tile map doesn't display.

Any help is appreciated.

kriskozak, 2015-11-16:
George, 
The static_transform_publisher that you included could be a problem.  The frame /wgs84 is a special frame that's handled directly by the transform_util.  I think that it is possible to override the name and treat /wgs84 as a standard /tf frame if it's in the tf tree, as it would be when you publish a static transform between it and map.

Georacer, 2015-11-16:
kriskozak,
I don't necessarily want to use /wgs84 for my own purposes. I'm just trying to make tiled map work, since I get the error message I described.
I was getting the same message with other local_xy_frame names.

Can you help me overlay a tiled map over my flying area? (Yes, I'm using a quadrotor)

kriskozak, 2015-11-16:
I'm just now also noticing that you have /local_xy_frame set as /wgs84 in the initialize origin node.  That should be /map. If you make both of those changes (also deleting the static transform publisher), do you still have the same issue?

Georacer, 2015-11-16:
So, you suggest that I only add this in my launch file:
```    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="crops"/>
        <rosparam param="local_xy_origins">
            [{ name: crops,
             latitude: 37.9879772,
             longitude: 23.9078602,
             altitude: 129.69,
             heading: 0.0}]
        </rosparam>
    </node>
```

I'll give it a go and let you know.

kriskozak, 2015-11-16:
George, 

Yes, that's right. However, I just tested it and was reminded of an issue.  As it currently works, /map needs to be in the /tf tree for transform_util to do the transform from /map to /wgs84, so you will probably also need to include a static transform publisher that goes between /map and some other frame. It doesn't matter what it is, it can be /void or /dummy, or something like that. This should work: 

    <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /dummy 100"  />

Also, I just brought up the tile map near your origin, and it looks like the imagery (MapQuest) is not very high resolution in your area :-(

Kris

Georacer, 2015-11-21:
Hello kriskozak,

I managed to make everything work with this launch file:
```
<launch>
    <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /dummy 100"  />

    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="Spata"/>
        <rosparam param="local_xy_origins">
            [{ name: Spata,
             latitude: 37.9879772,
             longitude: 23.9078602,
             altitude: 129.69,
             heading: 0.0}]
        </rosparam>
    </node>

    <node pkg="mapviz" type="mapviz" name="mapviz"/>
</launch>
```
Some notes, though:
On launch, I still get the "no tf from /wgs84 to /map" error. However, if I un-click and re-click the tickmark on the tile_map item, then the tf is generated and I get an "ok" message under the item.

Furthermore, I still need to drag the (road) map a bit for it to re-draw and render the display on my screen. I think this is a bug that needs attention.

Thanks for your help.

P.S. Since this issue was about lacking documentation, I'm not sure if it should be closed.

cbandera, 2015-11-21:
Hello,thank you for your detailed answers!
With your hints, I got it working, to the point, that mapviz receives my messages and transforms them. But the transformation itself seems to be skrewed. 
I am trying to transform a point fairly close to the local_xy_origin that I defined in the launch file. But after transformation of the lat/lon coordinates from the wgs84 to the map frame, they end up having really huge x/y coordinates: In the range of 10^6...

I guess these values are global UTM coordinates maybe? I would have expected them to be converted into the local_xy frame.

Heres my launch file:
```xml
<launch>
    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" output="screen">
        <param name="/local_xy_frame" value="/map"/>
        <!--Uncomment this, to use first GPS data for initialization-->
        <!--<remap from="/gps" to="/oxtsGPSData" />-->

        <!--Use the following to define a fixed coordinate for reference-->
        <param name="local_xy_origin" value="B3"/>
        <rosparam param="local_xy_origins">
            [{ name: B3,
            latitude: 49.0161447621,
            longitude: 8.49511214776,
            altitude: 115,
            heading: 0.0},
            ]
        </rosparam>
    </node>

    <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /dummy 100"/>

    <node pkg="mapviz" type="mapviz" name="mapviz" output="screen"/>
</launch>
```

closed 2015-11-21 by cbandera

malban, 2015-11-22:
The convention we have used for things in the /wgs84 frame is x=longitude and y=latitude