swri-robotics/mapviz #258: Build Issue

droter, 2015-10-26:
Hi,

After downloading the source from github and running the "rosdep install --from-paths src --ignore-src"
 command I am getting this error when I run catkin_make:

In file included from /home/matt/catkin_ws/src/marti_common/swri_image_util/include/swri_image_util/image_warp_util.h:46:0,
                 from /home/matt/catkin_ws/src/marti_common/swri_image_util/src/image_warp_util.cpp:30:
/usr/include/opencv2/legacy/legacy.hpp:1750:53: error: ‘cv::EM’ has not been declared
     CvEMParams( int nclusters, int cov_mat_type=cv::EM::COV_MAT_DIAGONAL,
                                                     ^
/usr/include/opencv2/legacy/legacy.hpp:1751:36: error: ‘cv::EM’ has not been declared
                 int start_step=cv::EM::START_AUTO_STEP,
-----
more of the same type of error
------
make[2]: *** [marti_common/swri_transform_util/CMakeFiles/swri_transform_util.dir/src/georeference.cpp.o] Error 1
make[1]: *** [marti_common/swri_transform_util/CMakeFiles/swri_transform_util.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** [marti_common/swri_image_util/CMakeFiles/swri_image_util.dir/src/image_warp_util.cpp.o] Error 1
make[1]: *** [marti_common/swri_image_util/CMakeFiles/swri_image_util.dir/all] Error 2
make: *** [all] Error 2

If I run "rosversion opencv2" I get: Cannot locate [opencv2]

It throws many errors after this spot.  Am I missing a dependency?

Thanks,

Matt

pjreed, 2015-10-26:
Which Ubuntu/ROS distributions and branch of marti_common are you using?  Check to make sure your git branches match your ROS distro.

ROS Hydro provided opencv2 as a ROS dependency, but as of Indigo it was removed and now it's preferred to depend on the system dependency libopencv-dev.  So, if you're using Hydro, try:
```
sudo apt-get install ros-hydro-opencv2
```
But if you're on Indigo, try:
```
sudo apt-get install libopencv-dev
```

droter, 2015-10-26:
I am using Ubuntu Description:	Ubuntu 14.04.3 LTS  Release:	14.04
with ROS indigo.  I ran the sudo apt-get install libopencv-dev and it said that it is already installed.  The branch I am using is ref: refs/heads/indigo-devel

pjreed, 2015-10-26:
It looks like the symbols the compiler is looking for should be defined in /usr/include/opencv2/ml/ml.hpp .  Check to see if you have that file, and if you don't, try:
```
libopencv-ml-dev
```

droter, 2015-10-27:
I have the file in  /usr/include/opencv2/ml/ml.hpp and libopencv-ml-dev is installed.  Do I need to set the rosversion opencv2 so it can find the directory?

pjreed, 2015-10-29:
I don't think so, I've been able to compile it on multiple different computers without doing that.  I suspect that there might be an issue somewhere with a missing transitive dependency due to something I have in my workspace that isn't in yours.

Just to verify -- do you have the mapviz, marti_common, and marti_messages repositories all in your workspace?  Are all of them on the indigo-devel branch?

shaun-edwards, 2015-10-29:
@pjreed, @droter, It appears mapviz is building correctly on the ROS servers [here](http://jenkins.ros.org/job/ros-indigo-mapviz_binarydeb_trusty_amd64/).  If there were a dependency issue, it is *almost* always caught there.

I've found that the package manager can mess things up sometimes.  One thing to try is to remove/install something that you know uses the openCV dependency.  You could try `cv_bridge` or if you want the nuclear option, remove/install `ros-indigo*`.  This seems extreme, but it doesn't take too long and it has fixed similar problems in the past for me.

droter, 2015-10-30:
Thanks for your help.  

There must be a conflict on my computer.  I installed the package on a clean install and it work perfect.

closed 2015-10-30 by droter