swri-robotics/mapviz #263: MapQuest disapearing and scale not working

davvdg, 2015-10-30:
Hi ! I'm new to both ROS and mapviz. I'm trying to load the mapquest map in the tile_map plugin, but I have trouble with it.
I've made a tf /wsg84 using the following command:
rosrun tf static_transform_publisher 0 0 0 0 0 0 /wgs84 map 1

Now, in mapviz, there is no projection message problem, and (sometimes) I am able to see the mapquest world map. But... it seems only the level 0 for tile 0 0 is being displayed regardless to the zoom I'm trying. Additionally, the red and green arrows of the origin cannot be kept in the center of the screen because it makes the whole map to disappear. The map only appear when the frame origin is placed either on the left or right of the screen and more often, when outside the top or bottom border of the screen... weird ?

Am I doing something wrong with the tf ? I'm not trying to display anything but the map, as I'm about to add a new tile_map provider for the maps made in my company.

David

malban, 2015-11-05:
davvdg,

First, you may want to look at the response in issue #256.

The issue is that there needs to be conversion lat/lon (wgs84) and the rest of the tf coordinates.  tf itself can't perform this conversion since it only supports translations and rotations.  The way mapviz handles going between global coordinates and tf coordinates is defining a lat/lon anchor point with respect to one of the tf frames and performing the projection of the lat/lon ellipsoid onto that plane.

If you take a look at the mapviz.launch file in the mapviz package you can see an example of this.  Running the launch file should allow you to visualize the map tiles correctly.  You may want to change the anchor point to a lat/lon closer to where your other data is.

Marc

pjreed, 2016-07-19:
There hasn't been any activity on this in over half a year, so I'm going to close it.

closed 2016-07-19 by pjreed