swri-robotics/mapviz #267: Documentation missing on TexturedMarker.resolution

Georacer, 2015-11-22:
Hello,

I've been using mapviz the last couple of days. Yes, documentation is lacking, but the content is awesome. Thanks for that!

I'm trying to work with TexturedMarker to project photos taken from a quadrotor to the ground.
To my understanding, the "resolution" field should affect the final image size, but the image should also scale when the altitude of the UAV changes, which doesn't happen.

That said, is there any hint on resizing the image by using the resolution field? Is there another, recommended method?

Thanks

malban, 2015-11-22:
The pose,  resolution, and frame_id fields determine how the image is projected onto the fixed frame in mapviz:
 * The resolution field represents the scale of a pixel in the image in meters.
 * The pose field represents the center of the image with respect to the frame_id in the header.

I think you could use the textured marker plug-in to visualize this data, but you will need to calculate the pixel resolution yourself based on the UAV height from the ground and the camera intrinsics.

You can also use the TexturedMarkerArray message in the same way as a MarkerArray and visualize multiple photos like a map.  One thing to keep in mind is that each image is loaded into texture memory on the GPU, so it's not a scalable solution.  

It should be possible in the future to make the plug-in more intelligent about how much to load and whether to automatically downsample based on the viewport, but there are no immediate plans to do so.

Georacer, 2015-11-22:
My question was more about scaling rather than the displayed image resolution itself.
Thanks for the answer.

closed 2015-11-22 by Georacer