swri-robotics/mapviz #276: Add Support for DeleteAll Markers

evenator, 2015-12-11:
The marker message recently added a new action type DELETEALL.

The constant is not available in [indigo](See http://docs.ros.org/indigo/api/visualization_msgs/html/msg/Marker.html), only [jade](http://docs.ros.org/jade/api/visualization_msgs/html/msg/Marker.html) but there's nothing to stop us from implementing the feature in Indigo by hardcoding a 3.

I already implemented this in the hydro branch, and I've found it pretty handy.

matt-attack, 2019-01-25:
@pjreed  is this resolved?

matt-attack, 2019-01-25:
It appears this has not been. Need to add an else if statement around here checking for an action of 3, and then clearing all markers if you catch it.

https://github.com/swri-robotics/mapviz/blob/master/mapviz_plugins/src/marker_plugin.cpp#L336

KindDragon, 2019-08-07:
Fixed in MR #609

closed 2019-08-07 by pjreed