swri-robotics/mapviz #280: Utilize ROS-I's Travis continuous integration (CI) scripts

shaun-edwards, 2015-12-30:
ROS-I now has some pretty generic scrips for utilizing Travis CI: https://github.com/ros-industrial/industrial_ci.

Travis CI allows checks to be performed on any PR made against the repo, see [here](https://github.com/ros-industrial/siemens_experimental/pull/8).

If there is interest, I can submit a PR against this and other swri repos to utilize this capability.

evenator, 2016-01-06:
@pjreed Have you seen this?

pjreed, 2016-01-06:
Ah, I had missed this while I was on break.  This looks neat.

We use TeamCity for CI internally, but it would be handy to have something external that we can integrate with Github.  I'll look into this when I have the chance.

malban, 2016-01-06:
Done

closed 2016-01-06 by malban