swri-robotics/mapviz #281: Segmentation fault when loading multires_image

kimmat, 2016-01-04:
Hi,

I get a Segmentation fault when loading a geo-file in mapviz. I have attached the gdb output in the file gdb.txt and the geo-file in kjeller.txt (renamed from kjeller.geo as github does not support uploading geo-files). I am using Ubuntu 14.04 and the following relevant packages:

ros-indigo-swri-transform-util (version: 0.0.7-0trusty-20151119-0320-+0000)
ros-indigo-mapviz (version: 0.0.3-0trusty-20151119-0324-+0000)

As far as I can see it fails the method swri_transform_util::GeoReference::Print(), but I don't know why. Any help is appreciated.

[gdb.txt](https://github.com/swri-robotics/mapviz/files/77220/gdb.txt)
[kjeller.txt](https://github.com/swri-robotics/mapviz/files/77223/kjeller.txt)

malban, 2016-01-04:
It looks like the cv::estimateRigidTransform function being used to calculate the transform may be expecting at least 3 points.  The quick fix would be to add a 3rd tie-point.  We will push an actual fix at some point.

Also, the projection field needs to be wgs84 or utm for mapviz to display it.  It could probably also be any valid tf frame I think.   I recommend using utm coordinates because that is what has been tested the most.

kimmat, 2016-01-05:
Thank you for you prompt response. Adding a third point fixed the issue. However I still have some problems.

I got another segmentation fault. I am attaching the gdb output and the configuration file (gdb2.txt and kjeller2.txt). In this setup I am only using one tile.

[gdb2.txt](https://github.com/swri-robotics/mapviz/files/78163/gdb2.txt)
[kjeller2.txt](https://github.com/swri-robotics/mapviz/files/78162/kjeller2.txt)

I also tried to use the original setup with two layers, but then I got an error saying that the images failed to load. I am attaching the error output (error3.txt), the configuration (kjeller3.txt) and the directory and file structure (tree3.txt).

[error3.txt](https://github.com/swri-robotics/mapviz/files/78177/error3.txt)
[kjeller3.txt](https://github.com/swri-robotics/mapviz/files/78179/kjeller3.txt)
[tree3.txt](https://github.com/swri-robotics/mapviz/files/78178/tree3.txt)

I think you have created a very nice visualization tool, and I hope you are able to help me so I am able to use it.

malban, 2016-01-05:
I just merged an existing pull request to fix the single layer segfault: #259

I think the issue with not loading the images is because of the png extension.  The current released version only supports jpg.  We did merge in support for other extensions recently, but it hasn't been released as a Ubuntu package yet.

kimmat, 2016-01-12:
I found one problem on my side. I had only one image in layer0, when it should be four (and had four in layer1, when it should have been one). Now it reports that everything is ok, but no image is shown. The Tile::Exist() method is called for all the images, but Tile::LoadImageToMemory is never called. I downloaded the latest version from github, and added some debugging output to find it out. Do you have any suggestions to where I should start looking for errors?

I am still using wgs84, since I do not know how to configure for utm. Utm usually have a zone/sector field, but there is no place to put that in the .geo file. Everything is working with the tile_map plugin, so I think the wgs84 configuration should be ok.

malban, 2016-01-12:
The UTM coordinates would be with respect to the zone that the images are in.  There is an assumption that this is within the same zone as the local origin defined in the launch file.

The code bases for displaying the tile_map and multires_image are different even though they do similar things.  multires_image was originally implemented with respect to UTM coordinates and there may be some bugs when using wgs84 coordinates since it hasn't received much attention yet.

My guess is that the issue is in the TileView class where it may be assuming the georeference parameters are in meters instead of degrees lat/lon.

kimmat, 2016-01-14:
I have been looking some more into the code, and I think the problem lies in the method GeoReference::GetTransform(). The method calculates a rigid transform between the image and the latlong-coordinates. I guess the method should convert the latlong coordinates into a local frame first, and then find the transform. Right now I get the transform:

[ INFO] [1452766822.273099052]: georeference:  transform: 0,000000, -0,000004, 59,978902
[ INFO] [1452766822.273406818]: georeference:             -0,000002, -0,000016, 11,069668
[ INFO] [1452766822.273695510]: georeference:             0,000000, 0,000000, 1,000000

I guess I could use the LocalXyFromWgs84 class to create the local frame, but then projection should be the tf frame name of the local origin not wgs84. Maybe GeoReference::GetTransform() should select the transform method based on datum, for instance datum could be either tf, wgs84 or utm depending on the representation of the tiepoints. But I think this will not work if the local origin moves, as then the translation in the transform should be updated.

I tried using a tf frame as the projection, and that worked. I did not manage to use utm and I don't know why, but I suspect that I have not setup the utm to tf transform correctly. I could find very little documentation on setting up the utm and wgs84 transforms, so I really don't know hoe this should be done.

kimmat, 2016-01-15:
I got the functionality I wanted using tile_map and making a local tile server (https://switch2osm.org/serving-tiles/building-a-tile-server-from-packages/). So I found a solution to my problem.

malban, 2016-01-15:
That's great.  I'm also planning on addressing the issues you found when time permits.