swri-robotics/mapviz #325: OpenGL blending is enabled but overwritten.

elliotjo, 2016-02-11:
OpenGL blending (alpha channel support) is enabled by default (in MapCanvas::initializeGL) but was broken by the support for QPainter.  It seems like there is some bad interaction with the call to QPainter::beignNativePainting() in MapCanvas::paintEvent().  Placing a call to glEnable(GL_BLEND) after the line restores functionality in plugins, but before does not.  This is just a work-around however, we should find the actual problem in case other GL state variables are being overwritten.

pjreed, 2016-02-22:
Are there any existing plugins in which this behavior can easily be demonstrated?

pjreed, 2016-03-23:
I believe that alpha transparency not working in the costmap plugin is a side effect of this.
@jgassaway @nick-alton

closed 2016-04-06 by pjreed