swri-robotics/mapviz #326: Image and Disparity Plugins Still Subscribed When Hidden

evenator, 2016-02-11:
Because these are high bandwidth message types, a user may want to reduce the load on the system by closing the subscription. Currently, the only ways to do that are to remove the plugin, which clears all of the settings, or to clear the topic selector, which is non-obvious and also requires the user to reselect the topic later.

Some plugins need to keep their subscribers open even when hidden because they keep a buffer of messages received. The image and disparity plugins only show the latest message, so they don't need to keep the subscription open when hidden.

Implementing unsubscribe-on-hide would require the mapviz plugin base class to provide OnHide() and OnShow() events that the plugin class can override. These might have other uses as well.

pjreed, 2016-02-11:
My suggestion would be to either make the SetVisible function virtual so that plugins can override it or provide show()/hide() slots similar to QWidget that are called when it is toggled.

pjreed, 2016-02-16:
Fixed in #328

closed 2016-02-16 by pjreed