swri-robotics/mapviz #330: Failed to send GPS data to mapviz

shengwen1997, 2016-02-21:
Hello, I have tried to publish my GPSFix data but mapviz showed me the following status:
"No message received"

And here is how I implement the publisher, can anyone give me the hint so I can fix the problem?
[Link](https://github.com/shengwen1997/ros_groundstation/blob/master/src/position_publisher.cpp)

Thanks

shengwen1997, 2016-02-21:
I also get the following message:
"[ WARN] [1456068927.408288043]: [transform_manager]: No transformer for transforming /far_field to /wgs84"

evenator, 2016-02-21:
Without digging too far into this, I suspect your problem is related to #256, so the discussion there might be helpful in solving your problem.

shengwen1997, 2016-02-22:
I replaced the topic "GPSFix" to "NavSatFix" then the problem is solved.

closed 2016-02-23 by evenator