swri-robotics/mapviz #364: /project/devel/lib/mapviz/mapviz': realloc(): invalid pointer:

layssi, 2016-05-31:
~/project$ roslaunch mapviz mapviz.launch 
... logging to /home/sam/.ros/log/3faa8890-2770-11e6-9179-3ca9f4125674/roslaunch-lap-014687-15595.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://lap-014687:40154/

SUMMARY
========

PARAMETERS
 * /initialize_origin/local_xy_frame: /far_field
 * /initialize_origin/local_xy_origin: swri
 * /initialize_origin/local_xy_origins: [{'latitude': 29....
 * /rosdistro: indigo
 * /rosversion: 1.11.19

NODES
  /
    initialize_origin (swri_transform_util/initialize_origin.py)
    mapviz_lap_014687_15595_1675689820302726453 (mapviz/mapviz)
    swri_transform (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[mapviz_lap_014687_15595_1675689820302726453-1]: started with pid [15613]
process[initialize_origin-2]: started with pid [15614]
process[swri_transform-3]: started with pid [15615]
*** Error in `/home/sam/project/devel/lib/mapviz/mapviz': realloc(): invalid pointer: 0x00007f144f55f840 ***

pjreed, 2016-06-02:
Hmm, I haven't seen that before... here's a few things to check:

- What's your OS?  Ubuntu 14.04?
- Do you have an old config file at `$HOME/.mapviz_config`?  If so, does removing it fix the problem?
- Does it still crash if you start a `roscore` separately and then run `rosrun mapviz mapviz` with no other nodes?

If it's still crashing, could you try to use gdb to get a stack trace and post that?

layssi, 2016-06-02:
I actually found the issue. It had something to do with RVIZ and openCV  being built with Qt5 support. This caused the camera display to crash. I rebuilt OPenCV without QT support and now it is working.

Thank you very much for your prompt response.

closed 2016-06-02 by layssi

Tiedan12138, 2024-01-12:
Hello, I have encountered the same problem. How did you solve it? Can you provide detailed steps