swri-robotics/mapviz #365: Cannot display Arrow marker

Georacer, 2016-06-06:
Hello!

I have trouble visualizing an arrow marker.

I think I have initialized everything properly, but I can't get it to work, even though I get an OK message in the topic (**marker/attractor**) subscription from Mapviz.

You can see the topic content and the Mapviz window in the attached picture.
[image]

fcu is the same frame as the rover, basically the body frame.

If I swap the marker type to 1 (a cube) then it displays correctly, which is even weirder.

I have tried

- passing all-zero pose and a 2-long points array
- passing only pose data
- both simultaneously

Can you replicate my issue or does it work for you?

Thank you in advance.

pjreed, 2016-06-07:
It looks as though support for the ARROW marker hasn't been implemented.  I'll work on that.

Georacer, 2016-06-07:
Thanks! Much appreciated. At least this time it wasn't me falling short on the API ;)

pjreed, 2016-06-08:
I've tentatively implemented support for arrows, although I don't have any bags with real data in them and am just using test data I've injected, so I might have missed some edge cases.  Feel free to try building my PR and see if they work for you.

PR #370 (2016-06-24): Implementing support for the ARROW marker type (Indigo)
Resolves #365
commit 5cb87dba (2016-06-24): Implementing support for the ARROW marker type

Resolves #365
commit ce5d5d6a (2016-06-24): Implementing support for the ARROW marker type

Resolves #365
commit 9f56a45e (2016-06-24): Implementing support for the ARROW marker type

Resolves #365
closed 2016-06-24 by malban