swri-robotics/mapviz #380: Failed to load image in multires_image

gondal1, 2016-06-28:
Hi, 
I have recently started working with mapviz and facing difficulty with multires_image and tile_map plug-ins. I have been able to run other plug-ins (gps, navsat, path, odometry, image) successfully. I have already gone through issue [#281](https://github.com/swri-robotics/mapviz/issues/281), but it doesn't seem to work for me. The status of multires_image is ``Failed to load image.`` Could you please point out where am I going wrong?

In tile_map plug-in, even though its status is OK, I am unable to see anything.
I am attaching my geo and launch files.  (attached files are text files)

[test_geo.txt](https://github.com/swri-robotics/mapviz/files/336987/test_geo.txt)
[launch.txt](https://github.com/swri-robotics/mapviz/files/336988/launch.txt)
The directory and file structure is
[tree.txt](https://github.com/swri-robotics/mapviz/files/336986/tree.txt)

The launch terminal shows repetition of following errors 

> [7effe3bffa40] [/mapviz_si_z0py9_13985_5837161233407603565/ImageCache::NetworkError:218]: NETWORK ERROR
[7effe3bffa40] [/mapviz_si_z0py9_13985_5837161233407603565/ImageCache::ProcessReply:194]: ============ AN ERROR OCCURRED ==============: http://otile1.mqcdn.com/tiles/1.0.0/sat/1/0/0.jpg
[7effe3bffa40] [/mapviz_si_z0py9_13985_5837161233407603565/ImagePtr tile_map::ImageCache::GetImage:148]: To many failures for image: http://otile1.mqcdn.com/tiles/1.0.0/sat/1/0/0.jpg
[7effe3bffa40] [/mapviz_si_z0py9_13985_5837161233407603565/ImagePtr tile_map::ImageCache::GetImage:148]: To many failures for image: http://otile1.mqcdn.com/tiles/1.0.0/sat/1/1/0.jpg
[7effe3bffa40] [/mapviz_si_z0py9_13985_5837161233407603565/ImagePtr tile_map::ImageCache::GetImage:148]: To many failures for image: http://otile1.mqcdn.com/tiles/1.0.0/sat/1/0/1.jpg
[7effe3bffa40] [/mapviz_si_z0py9_13985_5837161233407603565/ImagePtr tile_map::ImageCache::GetImage:148]: To many failures for image: http://otile1.mqcdn.com/tiles/1.0.0/sat/1/1/1.jpg

pjreed, 2016-06-28:
Are you able to see the map tiles (such as http://otile1.mqcdn.com/tiles/1.0.0/sat/1/0/0.jpg ) if you go to them directly in your web browser?

gondal1, 2016-06-28:
Yes, I am able to see them in my browser. Note: I am sitting under corporate proxy, could it be an issue? 
However, multires_image works now, but I can only use one image. I made the following change in geo file
> image_width: 512
image_height: 512
tile_size: 512

if I try to set image width and height 1024 then I get the previous error ``Failed to load image``. The directory and file structure that I use (with 1024 size) is attached above.Thanks

pjreed, 2016-06-28:
I think Mapviz is having an issue making the HTTP connection, so the corporate proxy could be an issue.  Do you know if it tries to redirect you to an SSL connection or change the certificate?  Do you have similar errors for other sources for `tile_map`?

I'm not very familiar with `multires_image`, but @malban might have some insight.

gondal1, 2016-06-28:
@NikolausDemmel could you comment on the SSL question? Thanks!

NikolausDemmel, 2016-06-29:
So as a quick workaround, I tried adding the following lines to the `ImageCache` constructor in the tile map plugin:

```
      QNetworkProxy proxy;
      proxy.setType(QNetworkProxy::HttpProxy);
      proxy.setHostName("localhost");
      proxy.setPort(3128);
      QNetworkProxy::setApplicationProxy(proxy);
```

I'm running a local cntlm proxy which manages the authentication to the company proxy, therefore "localhost" as proxy hostname.

With this, I was able to get some images loaded in mapviz. Unfortunately, after a short while, image started to fail again. It looks like this might be some rate-limiting of the company proxy. It seems like my IP address gets blocked for a short while by the company proxy, if I make too many requests. I'll have to investigate what the rules are. Maybe some rate-limiting of the image loading in the plugin implementation would help here. And it would also help if a failed image could be retried after some time, since the blocking by the parent proxy happens only for a short period of time, then connections are allowed again.

NikolausDemmel, 2016-06-29:
P.S.: All requests seem to be http, so I don't think this is an SSL issue.

NikolausDemmel, 2016-06-29:
Ah and I forgot to add that it would be great if someone could determine the proper place to setup the proxy (maybe in mapviz itself, not the individual plugins...). It would be great if it could read the proxy settings from the environment variables (on linux this is `http_proxy`, `https_proxy`, `ftp_proxy`, `socks_proxy`, `no_proxy`) and use those to set up the proxy to use. On top of that there could be ros parameters that overwrite the environment variables with some manual settings.

pjreed, 2016-06-29:
Ah, that would do it.  Mapviz currently doesn't have any kind of proxy support, so if your corporate proxy isn't transparent, it won't be able to connect through it.  

Unfortunately, there's not a standard, cross-platform way of managing proxies.  We could try reading from the environment to get default values, but it might be better to add a panel to Mapviz that can be used to configure application-wide proxies.

NikolausDemmel, 2016-06-29:
> Unfortunately, there's not a standard, cross-platform way of managing proxies. We could try reading from the environment to get default values, but it might be better to add a panel to Mapviz that can be used to configure application-wide proxies.

True. I was just thinking of using the env variables at least as a default would be a nice experience, since on Linux many tools already pick up proxy information through those. Manual specification of the proxy through config or GUI would maybe be the best thing to add first. Automatic detection can be added later for the highly used cases (like those env variables on Linux).

paulhodel, 2016-06-30:
pjreed can I have your email to discuss a private matter?

Pkumar93, 2018-04-03:
Hello!

I am having the same issue and was hoping someone could help me out. I have been able to use the tile map from Daniel snider to load google map satellite images. But I would like to make the multires image work as it wouldn't always be possible for me to be connected to the internet to load new unexplored areas.

I am trying to get the multires image plugin to work. From my understanding the .geo is some sort of config file and there needs to be an array of folders and images to display tiles of different zoom scale?

I have tried to use the GMapCatcher from https://github.com/heldersepu/gmapcatcher, but the maps do not load. 

Also i cannot see the tile maps on http://otile1.mqcdn.com/tiles/1.0.0/sat/1/0/0.jpg and the site fails to load. I would really appreciate some help on this. A small list of to do's in order to download the images would be nice. Since the GMapCatcher fails to load.

Thank you.

malban, 2018-04-03:
The tile_map plugin can be used to display tiles from disk if you have downloaded them with a tool.  Take a look at this issue: #575 

To load tiles locally and bypass the network you would select custom source in the tile_map plugin and set the base url to something like: file:///home/malban/map_tiles/{level}/{x}/{y}.png

Pkumar93, 2018-04-03:
I think there is something fundamentally wrong with my .geo file. I am not certain as to how i should enter the tie points values. Any help would be appreciated. This is my first project dealing with maps and i am struggling to get this part part running. Running offline is fairly crucial to my project

malban, 2018-04-03:
I would recommend using tile_map instead of multires_image.  tile_map can be used offline.  They provide essentially the same functionality, but tile_map is easier to use in terms of getting data.  See #575 for an example of someone using it offline.