swri-robotics/mapviz #381: Changing "heading" in "local_xy_origins" doesn't make any difference.

gondal1, 2016-06-29:
Following the explanation in [#256](https://github.com/swri-robotics/mapviz/issues/256), I fixed the origin of the 'map' frame to a geographic coordinate in ``local_xy_origins``. 

>     <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 48.891690,
         longitude: 9.172950,
         altitude: 306.471004486,
         heading: 0.0}]
    </rosparam>

I tried changing ``heading`` to different values but nothing worked. Running `lat_lon_tf_echo` node with fixed_frame and target_frame set to `map` always returns heading to be 90°.

> Latitude: 48.891690°, Longitude: 9.172950°, Heading: 90.000000°

However, if I introduce an offset in``transform_util.cpp``'s ``ToYaw`` method then I see the corresponding change, but I guess this is not the right way to do it. Any suggestions about that?

gondal1, 2016-06-30:
In  `initialize_origin.py`, I found out that `_gps_fix.track` is fixed to 90°. I changed it to `_gps_fix.track = origin["heading"]` which seems to fix this issue. This corrects my path messages' transformations ( which contains lat / lon data), but as a result of this my `multires_image` also gets rotated which is not very convenient.  Could you explain why is this happening? Is there any way to introduce `heading` only for your path messages transformation, and avoid its affect on tile_map and multires_image? Thanks!

Note: My GPS position gets displayed correctly in every case.

pjreed, 2016-06-30:
Well, that sort of makes sense.  Changing the heading on the local_xy_origin would effectively be rotating the rest of your tf tree with respect to the wgs84 coordinate frame.  How is it not working properly when the heading is set to 0.0?

gondal1, 2016-07-01:
Yes, you are right. Instead of changing the heading, I needed to introduce a static transform between map and my path messages frames (). So, I have reverted the changes in  `initialize_origin.py ` and changed   `heading ` back to 0.0. It works now. Thanks a lot!

closed 2016-07-01 by gondal1