swri-robotics/mapviz #391: doesnt compile for kinetic

davidrotor, 2016-07-04:
catkin_make install mapviz failed 

/usr/include/geos/platform.h:66:26: error: conflicting declaration ‘typedef long long int int64’
    typedef long long int int64;
                          ^
In file included from /usr/local/include/opencv2/core/cvdef.h:67:0,
                 from /usr/local/include/opencv2/core.hpp:52,
                 from /usr/local/include/opencv2/core/core.hpp:48,
                 from /home/david/Paquetes/src/marti_common/swri_geometry_util/include/swri_geometry_util/intersection.h:31,
                 from /home/david/Paquetes/src/marti_common/swri_geometry_util/src/intersection.cpp:32:
/usr/local/include/opencv2/core/hal/interface.h:57:20: note: previous declaration as ‘typedef int64_t int64’
    typedef int64_t int64;

evenator, 2016-07-05:
What branch of marti_common are you using? This should be fixed in `kinetic-devel`.

davidrotor, 2016-07-05:
ok I changed to kinetic devel and still get

from
/home/david/Paquetes/src/marti_common/swri_geometry_util/src/intersection.cpp:34:
/usr/include/geos/platform.h:66:26: error: conflicting declaration ‘typedef
long long int int64’
    typedef long long int int64;
                          ^

davidrotor, 2016-07-05:
I fixed the typedef manually but keep getting this error

/usr/bin/ld: cannot find -linux-gnugeos-3.5.0

I only know I do have libgeos-3.5.0 installed

evenator, 2016-07-05:
That problem should be fixed in kinetic-devel. Can you do a `catkin_make clean` on your workspace and rebuild? I'm thinking that CMake left some bad configuration around from before you checked out kinetic-devel.

davidrotor, 2016-07-05:
yeah just fixed it thank you Edward!

closed 2016-07-06 by evenator