swri-robotics/mapviz #392: Adding a Multi Res Image

neweirik, 2016-07-06:
Hi, I'm trying to add a single image tile underneath my laser scan and robot model. I got everything working except the map. I have a map called tile00000x00000.png inside a folder called layer0. Both files are in the same folder as the .geo file: 

image_path: "." 
image_width: 500     
image_height: 500    
tile_size: 500         
extension: "png"

datum: "wgs84"        
projection: "utm"         
                 
tiepoints:             
 - point: [250,250, 374737.32, 12285.28] 
 - point: [1,1, 37519.36, 1221221.24]

I have tried jpg and png, but I only get this error message:

[ INFO] [1467806775.074415033]: Accept multires image configuration.
[ INFO] [1467806775.074971120]: georeference: Image path is /home/ugv/Desktop/test_kart/.
[ERROR] [1467806775.075186194]: yaml-cpp: error at line 0, column 0: bad conversion
[ERROR] [1467806775.075258367]: Error: Failed to load image.

Is this a known issue? Or am I doing something wrong.

malban, 2016-07-06:
The error looks to be a yaml parsing problem.  I'll see if I can reproduce it.  Also, I think there needs to be 3 tiepoints.

neweirik, 2016-07-06:
I got the same error with 3 points

malban, 2016-07-06:
The yaml you have there looks correct, but that error usually indicates one of the necessary fields is missing.  It's possible the text encoding of the file is causing a problem.  

I've committed a change to the marti_common repository to improve the warning log messages when there is a parse error with the georeference file.

I've also attached an example .geo file which parses correctly for me.

[example.geo.txt](https://github.com/swri-robotics/mapviz/files/350722/example.geo.txt)

neweirik, 2016-07-07:
Thank you! With the new commit I was able to get the green "status: ok" for the map, but Im having some trouble with the tf frames. When using tf from the [turtlesim tf broadcaster](http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20broadcaster%20%28C%2B%2B%29) I get this output: 
[ INFO] [1467876431.249816435]: georeference:  width = 500
[ INFO] [1467876431.249884609]: georeference:  height = 500
[ INFO] [1467876431.249999429]: georeference:  tile_size = 500
[ INFO] [1467876431.250067922]: georeference:  extension = jpg
[ INFO] [1467876431.250133280]: georeference:  datum = world
[ INFO] [1467876431.250183073]: georeference:  projection = world
[ INFO] [1467876431.250266855]: georeference:  tiepoints
[ INFO] [1467876431.250348364]: georeference:     [0, 0, 680146.200000, 4763853.400000]
[ INFO] [1467876431.250400995]: georeference:     [499, 0, 680500.500000, 4763872.800000]
[ INFO] [1467876431.250461621]: georeference:     [499, 499, 680515.300000, 4763458.700000]
[ INFO] [1467876431.250494801]: georeference:  transform: 0.710020, 0.029659, 680146.200006
[ INFO] [1467876431.250521103]: georeference:             0.038878, -0.829860, 4763853.400013
[ INFO] [1467876431.250546160]: georeference:             0.000000, 0.000000, 1.000000
QThread::setPriority: Cannot set priority, thread is not running
QThread::setPriority: Cannot set priority, thread is not running
[ WARN] [1467876431.328224576]: Wgs84Transformer not initialized
[ WARN] [1467876455.181042205]: Wgs84Transformer not initialized

Somehow the Wgs84Transformer is not initialized. Any idea on how to fix this?

closed 2016-07-08 by malban