swri-robotics/mapviz #401: Cant launch Mapviz con Kinetic

davidrotor, 2016-07-12:
I installed the new mapviz kinetic branch and when I roslaunch it

... logging to /home/david/.ros/log/bbaff856-4879-11e6-ad23-dc85de528312/roslaunch-david-31454.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt
Done checking log file disk usage. Usage is <1GB.

started roslaunch server http://david:43677/

SUMMARY
========

PARAMETERS
 * /initialize_origin/local_xy_frame: /far_field
 * /initialize_origin/local_xy_origin: swri
 * /initialize_origin/local_xy_origins: [{'latitude': 29....
 * /rosdistro: kinetic
 * /rosversion: 1.12.2

NODES
  /
    initialize_origin (swri_transform_util/initialize_origin.py)
    mapviz_david_31454_321262999549899683 (mapviz/mapviz)
    swri_transform (tf/static_transform_publisher)

ROS_MASTER_URI=http://localhost:11311

core service [/rosout] found
process[mapviz_david_31454_321262999549899683-1]: started with pid [31472]
process[initialize_origin-2]: started with pid [31473]
process[swri_transform-3]: started with pid [31474]
*** Error in `/home/david/Paquetes/install/lib/mapviz/mapviz': realloc(): invalid pointer: 0x00007f3da66ce820 ***
======= Backtrace: =========
/lib/x86_64-linux-gnu/libc.so.6(+0x77725)[0x7f3d9f4a7725]
/lib/x86_64-linux-gnu/libc.so.6(realloc+0x348)[0x7f3d9f4b3f88]
/usr/lib/nvidia-340/tls/libnvidia-tls.so.340.96(+0x214a)[0x7f3d9cdec14a]
======= Memory map: ========

evenator, 2016-07-13:
We're not using kinetic on any of our internal projects, so I only minimally tested the kinetic branch before releasing it. It's entirely likely that there are some bugs in there. It looks like you're building from source, are you using the `kinetic-devel` branch of `marti_common` as well?

davidrotor, 2016-07-13:
Yes both as kinetic

pjreed, 2016-07-13:
For what it's worth, I've been able to successfully build and launch mapviz in kinetic.  I only checked out the mapviz repository and used rosdep to install all of its dependencies, so I didn't build marti_common or marti_messages.

It's a little suspicious that the backtrace goes into libnvidia-tls.so.  Can you run other OpenGL programs like `rviz` successfully?  Have you tried upgrading to a newer version of the NVIDIA driver?

davidrotor, 2016-07-14:
yes I have my rviz running and rqt. I will try the nvidia driver upgrade
today ill keep you posted

pjreed, 2016-12-22:
There's been no activity on this for a while and I know other people are using mapviz on Kinetic, so I'm going to close this.

closed 2016-12-22 by pjreed