swri-robotics/mapviz #402: Cant load tiles from mapquest

davidrotor, 2016-07-18:
how can I load tiles from other sources if these are deprecated?
[image]

pjreed, 2016-07-18:
For what it's worth, the Stamen map sources listed in the `tile_map` tool still work, although they're a bit lower quality.

I'm also working on some modifications right now that will make it easier to add other map sources to the tool, and hopefully that'll be ready soon.

davidrotor, 2016-07-18:
Ty ill use stamen map for now and wait for update!, how do I add a marker
to the map?

pjreed, 2016-07-18:
To display markers on the map, you'll want to publish them on a ROS topic as a [visualization_msgs/Marker](http://docs.ros.org/api/visualization_msgs/html/msg/Marker.html) message and then add a `marker` plugin in Mapviz for that topic.

davidrotor, 2016-07-18:
If i want to create markers from clicking in the map is it posible?

pjreed, 2016-07-19:
Unfortunately, it's not currently possible to draw markers directly on the map.  It sure would be nice to have, though. :smile_cat: I'll add that to the list of things to do...

PR #403 (2016-07-20): Giving tile_map an interface overhaul (Indigo)
MapQuest has turned off their public API for map tiles, so this plugin needed some work.  I have:
- Removed the MapQuest sources
- Made the interface for adding new sources more powerful
- Overhauled how sources are saved and loaded under the hood

Resolves #402 

To test this, try adding ArcGIS's official demo server.
1. Select `Custom...` as the `Source`
2. Set the `Base URL` to `https://server.arcgisonline.com/ArcGIS/rest/services/World_Imagery/MapServer/tile/`
3. Clear the `Suffix` (it should be empty)
4. Set the `Coord order` to `z/y/x`
5. Se [... cut]
commit 0682b70b (2016-07-20): Giving `tile_map` an interface overhaul

MapQuest has turned off their public API for map tiles, so this plugin needed some work.  I have:
- Removed the MapQuest sources
- Made the interface for adding new sources more powerful
- Overhauled how sources are saved and loaded under the hood
- Added a button to reset the current tile cache

Resolves #402
PR #404 (2016-07-20): Giving tile_map an interface overhaul (Jade)
MapQuest has turned off their public API for map tiles, so this plugin needed some work.  I have:
- Removed the MapQuest sources
- Made the interface for adding new sources more powerful
- Overhauled how sources are saved and loaded under the hood
- Added a button to reset the current tile cache

Resolves #402
commit 9a2980c6 (2016-07-20): Giving `tile_map` an interface overhaul

MapQuest has turned off their public API for map tiles, so this plugin needed some work.  I have:
- Removed the MapQuest sources
- Made the interface for adding new sources more powerful
- Overhauled how sources are saved and loaded under the hood
- Added a button to reset the current tile cache

Resolves #402
closed 2016-07-20 by malban