swri-robotics/mapviz #421: tile_map: Maps not reliably loading

jhoare, 2016-08-24:
When attempting to load mapviz using a tile_map, it does not load every square of the map. Sometimes all (or at least most) of the squares will load, but most of the time it is very patchy and it seems like it has given up loading. 

I do sometimes get a "Too Many Failures.." message to the console, but I often don't get any errors printed to the console. 

I'm mostly working with the Bing Maps source but i seem to get similar results with the stamen sources as well. 

I wonder if this too is related to my 4k display, and so maybe many more tiles are being requested than you guys work with?

pjreed, 2016-08-24:
I think this is the same as #342.

I don't think this is specifically because of a 4k display, but it may exacerbate the problem.  Basically, when it tries to load a lot of tiles at once, some of the HTTP requests may time out or fail, and currently the `tile_map` plugin doesn't retry failed connections; it really should.

pjreed, 2016-09-30:
Closing since this was fixed in #423.

closed 2016-09-30 by pjreed