swri-robotics/mapviz #430: Support of image_transport compressed datatypes

jhoare, 2016-09-19:
We would like to use the compressed image formats so we can more efficiently stream images over a wireless network.

pjreed, 2016-09-19:
Mapviz's `image` plugin already uses ImageTransport under the hood to handle it subscriptions, which means it is possible to make it used a compressed image transport by setting the appropriate parameter.  For example, if mapviz is running as a node named `mapviz_1474307178780514021`, you can run this:
```
$ rosparam set /mapviz_1474307178780514021/image_transport compressed
```

Afterward, mapviz will subscribe to the compressed versions of any image topics that support it.

It would be nice if this could be configured through the GUI, though.

jhoare, 2016-09-19:
Ah okay good to know. I was expecting the interface to be more similar to rqt_image_view where I select the "compressed" image topic to used the compressed transport.

jhoare, 2016-09-19:
Note, when using "compressed" image transport and visualizing multiple images I get the following error:

```
[7fe9d06c3a40] [/commandcenter/mapviz_messier_20127_7398547913354522751/ServiceManager::advertiseService:147]: Tried to advertise a service that is already advertised in this node [/commandcenter/mapviz_messier_20127_7398547913354522751/compressed/set_parameters]
```

pjreed, 2016-09-19:
Hmm, I can confirm that, and it is peculiar.  It doesn't seem like that hurts anything, but it's not immediately obvious why ServiceManager::advertiseService would end up getting called as a result of ImageTopic::subscribe... I'll investigate.

evenator, 2016-09-19:
ImageTransport::subscribe is a little...magical. For the compressed topic, it creates a service for communicating the compression information from the publisher to the subscriber. Other image transport plugins have other problems--Theora cannot handle multiple subscribers or resubscribing. Those are the only two I've used, so I can't comment further, but suffice to say, ImageTransport can be a little fragile when you're using it for anything besides single publisher->single subscriber as in a processing pipeline.

PR #432 (2016-10-06): Add a GUI for controlling the Image Transport (Indigo)
This will add a sub-menu under the "View" menu that will:
- List all available image transports
- Indicate which one is currently the default
- Allow the user to choose which one will be used for new ImageTransport subscriptions
- Save and restore this setting to Mapviz's config file
- Cause any `image` plugins using the default transport to resubscribe

In addition, the `image` plugin now has a menu that can be used to change the transport for that specific plugin so that it is different from the default.

Fixes #430
commit eaecdedb (2016-10-06): Add a GUI for controlling the Image Transport (#432)

This will add a sub-menu under the "View" menu that will:
- List all available image transports
- Indicate which one is currently the default
- Allow the user to choose which one will be used for new ImageTransport subscriptions
- Save and restore this setting to Mapviz's config file
- Cause any `image` plugins using the default transport to resubscribe

In addition, the image plugin now has a menu that can be used to change the
transport for that specific plugin so that it is different from the default.

Fixes #430
PR #436 (2016-10-06): Add a GUI for controlling the Image Transport (Jade)
This will add a sub-menu under the "View" menu that will:
- List all available image transports
- Indicate which one is currently the default
- Allow the user to choose which one will be used for new ImageTransport subscriptions
- Save and restore this setting to Mapviz's config file
- Cause any `image` plugins using the default transport to resubscribe

In addition, the image plugin now has a menu that can be used to change the
transport for that specific plugin so that it is different from the default.

Fixes #430
commit a6115ee9 (2016-10-06): Add a GUI for controlling the Image Transport (#432)

This will add a sub-menu under the "View" menu that will:
- List all available image transports
- Indicate which one is currently the default
- Allow the user to choose which one will be used for new ImageTransport subscriptions
- Save and restore this setting to Mapviz's config file
- Cause any `image` plugins using the default transport to resubscribe

In addition, the image plugin now has a menu that can be used to change the
transport for that specific plugin so that it is different from the default.

Fixes #430
closed 2016-10-06 by evenator