swri-robotics/mapviz #431: local WMTS for tile_map plugin problems

ganhao89, 2016-09-29:
I am trying to build a local tile map server for the tile_map plugin, so I can navigate and show my husky robot in the map at where no internet is available. I followed the link here to build a tile map server:

https://www.linuxbabe.com/linux-server/openstreetmap-tile-server-ubuntu-16-04

I was able to see my map in a browser when typing 'http://my-server-ipt' after following the instructions in the link above. I was also able to see the tile of a world map in the browser when typing 'http://localhost/osm_tiles/0/0/0.png'. Everything described in the above link works! 
The problem is, when I launch the mapviz.launch, select 'tile_map' plugin and choose 'Custom WMTS Source', in the Base URL, type 'http://localhost/osm_tiles/{level}/{x}/{y}.png', nothing shows. I tried change the first part of Base URL into http://localhost, http://127.0.0.1/osm_tiles and so on, none of them works.
I am thinking maybe I need to do some additional steps after following the above link? I am new to the map server thing, please help!
Thanks,
Hao

pjreed, 2016-09-30:
At a glance, that seems like it should be correct. A few things to check:
1. Did you click "Save" after entering the custom WMTS URL?
2. Have you tried closing and re-opening Mapviz?  (sounds silly, but I have actually seen it not load tiles immediately after adding a new URL a couple of times...)
3. Does mapviz print any warning or error messages that look relevant to the console?

I'm not personally familiar with setting up OSM, but I've fairly often used Mapviz together with [MapProxy](https://pypi.python.org/pypi/MapProxy) to create tile caches from other map servers.

ganhao89, 2016-09-30:
No, there is no warning message. But I haven't tried "Save" or re-opennig Mapviz. I will give it a try today to see if it works. Thanks.

pjreed, 2016-10-03:
Clicking "Save" is necessary in order to make it actually start loading tiles for the new server, so that is probably what was missing.  I should probably make that a little more obvious....

danthony06, 2026-09-20:
https://github.com/swri-robotics/mapviz/pull/929 Adds a lot of UI feedback, documentation, and utilities for improving the workflow. If there are more specific suggestions, please open a new issue.

closed 2026-09-20 by danthony06