swri-robotics/mapviz #440: tilemap cannot be released for jade

evenator, 2016-10-24:
Could not resolve rosdep key 'libjsoncpp' for distro 'utopic':
No definition of [libjsoncpp] for OS version [utopic]
	rosdep key : libjsoncpp
	OS name    : ubuntu
	OS version : utopic
	Data: fedora:
- jsoncpp
ubuntu:
  precise:
  - libjsoncpp0
  saucy:
  - libjsoncpp0
  trusty:
  - libjsoncpp0
  vivid:
  - libjsoncpp0
  wily:
  - libjsoncpp0v5
  xenial:
  - libjsoncpp1

evenator, 2016-10-24:
Ditto for xenial

pjreed, 2016-10-24:
Well, that's unfortunate.  https://raw.githubusercontent.com/ros/rosdistro/master/rosdep/base.yaml has the following for jsoncpp-related things:

```yaml
libjsoncpp:
  fedora: [jsoncpp]
  ubuntu:
    precise: [libjsoncpp0]
    saucy: [libjsoncpp0]
    trusty: [libjsoncpp0]
    vivid: [libjsoncpp0]
    wily: [libjsoncpp0v5]
    xenial: [libjsoncpp1]
libjsoncpp-dev:
  arch: [jsoncpp]
  debian: [libjsoncpp-dev]
  fedora: [jsoncpp-devel]
  gentoo:
    portage:
      packages: [dev-libs/jsoncpp]
  ubuntu: [libjsoncpp-dev]
```

So libjsoncpp is indeed missing on utopic, although it looks like it's present for xenial.

pjreed, 2016-10-24:
@evenator , give it another try now.

pjreed, 2017-05-10:
Is this still an issue or can it be closed?

evenator, 2017-05-10:
I don't know. I' have to try it again.

evenator, 2017-05-13:
Confirmed that tile_map is building in jade: http://build.ros.org/job/Jbin_uT64__tile_map__ubuntu_trusty_amd64__binary/

closed 2017-05-13 by evenator