swri-robotics/mapviz #441: Multi-res images questions: where to download the geo-referenced maps

ganhao89, 2016-11-09:
Does anybody know where I can download multi-res tiled images for the multi-res image plugin? 
I have successfully build a local tile Open Street Map (OSM) server for the Tile Map plugin. Now I need satellite images instead of OSM. And I need to use it offline. However, I couldn't find anywhere to download the geo-referenced maps. So if anyone knows 
1. Where i can download geo-referenced satellite maps (need to purchase is fine)
2. How to set up after downloading the maps
Please help.

Thanks!

evenator, 2016-11-22:
You might want to look at https://mapproxy.org/

malban, 2016-11-22:
The multi-res image plugin is a custom format so there is no convenient source online.   Existing or scraped image data can be used by putting it in the correct directory structure with a georeference file.  A much more convenient method for offline maps is to use something like mapproxy and tile_map.  We'll add some more detailed documentation along those lines.

vincentrou, 2017-01-31:
Here is a script that I use to download tile and use as multires_image

https://github.com/vincentrou/gmapcatcher/blob/6f9e71d1fd418335497d168b8b3165aa0190830a/downloadTileTop25.py

danielsnider, 2017-02-11:
@vincentrou thanks for the script! can you provide a .geo file too?

I'm looking for full example of using multires_image with a .geo file and folder structure with images.

danielsnider, 2017-05-11:
Btw, to help myself with this issue I created:
https://github.com/danielsnider/MapViz-Tile-Map-Google-Maps-Satellite

tldr: you can put google maps satellite view in mapviz like this
```
$ sudo docker run -p 8080:8080 -d -t -v ~/mapproxy:/mapproxy danielsnider/mapproxy
Then put this URL into the MapViz option "Custom WMTS Source..." http://localhost:8080/wmts/gm_layer/gm_grid/{level}/{x}/{y}.png
```
Maybe this will help others.

facontidavide, 2017-10-05:
@danielsnider It sure helped me

closed 2021-06-11 by danthony06