swri-robotics/mapviz #469: Recentre of viewport of map on vehicle after draging the map away from the centre

iravi89, 2016-12-30:
Hi,
Can you please help in doubt which i am fail to solve,
As i can visualize the gps data on map and tracking of vehicle but i have doubt how it will recenter itself to the center if i drag the map away from center .
how i can change or handle the viewport.

pjreed, 2017-01-03:
I'm a little unclear on exactly what you need; if you want to re-center mapviz on your target frame's origin after you have dragged away from it, you can click the icon in the bottom-right corner of the window:
[image]

iravi89, 2017-01-04:
Hi PJ
Great Thanks for your reply,And what you said i have tested but problem is i need to  have continuously update the origin of target frame and when i press the reset button it just move viewport to initial position from where the target frame has started.

Please guide me how i can update the origin of the target frame on every new position.

Thanks