swri-robotics/mapviz #477: How to get the geo file?

YuxiHuang001, 2017-02-18:
Hi, I a beginner to mapviz, and I'm trying to show the satellite map like the video shown in mapviz wiki, I think I need to use "Multi-res Image" Plug-in unit to load geo file to show the satellite map in mapviz. So where can I get the geo file? Does it for free download? Thx a lot

evenator, 2017-02-18:
As of this time, we don't have a good tool available for making geo files. See #441 

Some intrepid mapviz users have come up with their own solutions. Daniel Snider created [this tools](https://discourse.ros.org/t/easy-offline-google-maps-layer-for-mapviz/1317) to use the tilemap plugin when you're offline. Vincent Rou's [GMap Catcher](https://github.com/vincentrou/gmapcatcher) can also output MapViz multires image geo files.

closed 2017-02-18 by evenator

YuxiHuang001, 2017-02-19:
Ok, I will have a try, thank you very much~