swri-robotics/mapviz #478: point_click_publisher to publish lat/lon coordinates of tile map

ramboto, 2017-04-21:
Hello,
I have a tile map running on mapviz and I used the initialize_origin.py tool to initialize a lat/lon position on my robot's local frame. I can see the tile map and as I move the mouse around I can see both lan/lon numbers changing and the robot frame xy changing.

I'm trying to publish lat/lon coordinates using point_click_publisher whenever I click on the map, however I'm not sure what frame to use for the tile map. Whenever I click on the map, I'm just publishing the robot's local frame. Is there a tf transform I need to perform first?

ramboto, 2017-04-21:
I was looking in the local_xy_utility.cpp (part of swri_transform_util) and saw that I can use the function Wgs84FromLocalXy to get the transformation. So it looks like I can subscribe to the point_click_publisher and call that function to get the transform. My guess is this the way I should go about it.

malban, 2017-04-21:
The easiest way would be to type /wgs84 in the Frame dropdown for the point click plugin

danielsnider, 2017-05-11:
@malban how do I make the /wgs84 frame available? thanks!

malban, 2017-05-11:
@danielsnider 

The /wgs84 frame is not a true tf frame (since it can't be defined with a simple translation and rotation), so it won't be available through the standard tf interfaces.  For the point_click_publisher you would just type /wgs84 into the frame dropdown box.  This will work as long as the /local_xy_origin has been published and the tf frame it is associated with is connected to the frame mapviz is displaying.

I see in your launch file that the initialize origin node is set to 'auto' and will only publish the origin once it receives a GPS message.  Can you confirm that the /local_xy_origin topic has been populated?  If not, you can set it to one of the manually defined origins instead of 'auto'.

To use the /wgs84 frame in code as if it were a tf frame you would use swri_transform_util::TransformManager in place of a tf::TransformListener and use swri_transform_util::Transform instead of tf::Transform.  The transform manger wraps the standard tf listener and also listens for the local xy orign so it can perform the conversion to and from lat / lon and 
 frames in the tf tree.

malban, 2017-05-11:
Closing issue.

closed 2017-05-11 by malban

danielsnider, 2017-05-12:
Thanks for the super helpful response! I am still having trouble. Can you look at my question on [answers.ros.org/question/261580/how-do-i-to-make-wgs84-frame-available/](http://answers.ros.org/question/261580/how-do-i-to-make-wgs84-frame-available/)?

pjreed, 2017-05-12:
I tried to make a post there, but it's awaiting moderation.  Here's a copy of it:

> "/wgs84" isn't a real TF frame and isn't handled by the normal ROS TF system, so it won't show up in the normal TF tree. It's handled specially by the swri_transform_util package.
> 
> You could try putting a forward slash in front of it in the plugin panel -- so "/wgs84" rather than "wgs84" -- but I don't think that should make a difference. Do you have a topic at "/local_xy_origin"? What does echoing it produce? (it should have been published by initialize_origin.py after I got its first NavSatFix)

danielsnider, 2017-05-12:
I have the topic "/local_xy_origin" and it gives me lat long GPS of my origin when it was initialized by initialize_origin.py. 

What could potentially be going wrong for wgs84 to still not work? Any other ideas?

Why was an explicit wgs84 frame on the tf tree not implemented? Out of curiosity.

Thanks!

> On May 12, 2017, at 9:55 AM, P. J. Reed <notifications@github.com> wrote:
> 
> "/local_xy_origin"

pjreed, 2017-05-12:
ROS's TF mechanism is only capable of using translation and rotation to transform between frames.  Since WGS84 coordinates map to a roughly-spherical surface, it's not possible to map between a 2D coordinate grid and that using only translation and rotation.  `swri_transform_util` has some methods that act as wrappers around the standard TF methods but do the right magic to make the `/wgs84` frame work properly.

Actually, I just tested it in my mapviz, and I can confirm that setting the frame to `wgs84` does not work but `/wgs84` does work.  Could you see if adding the forward slash makes a difference for you?

malban, 2017-05-12:
We should probably fix swri_transform_util to correctly handle the non forward slash case

danielsnider, 2017-05-13:
Confirmed that was my problem, I had missed the slash in "/wgs84". Thank you!