swri-robotics/mapviz #483: Add ability to set /local_xy_origin in Mapviz

pjreed, 2017-05-11:
Normally the origin is set by the `swri_transform_util/initialize_origin.py` node, but if that node isn't running or isn't initialized properly, the /wgs84 frame won't work right and Mapviz won't be able to do things like render tile maps.

It would be handy if Mapviz could detect whether that has been initialized properly, display a warning to the user if it hasn't been, and provide the ability to set it to a fixed point.

facontidavide, 2017-11-10:
I can work on this, since it is giving me some headache...

pjreed, 2017-11-10:
Sure, feel free.  I haven't worked on it mostly because for my purposes, it's easy to just start a [swri_transform_util/initialize_origin.py](https://github.com/swri-robotics/marti_common/blob/master/swri_transform_util/nodes/initialize_origin.py) node and let it take care of it, but it would be convenient to have a way to notify the user if it hasn't been set and allow them to manually set it.

facontidavide, 2017-11-11:
I got something that works quite well, at least for me (famous last words).
It is still a little rough around the edges, though.

https://github.com/Eurecat/mapviz/tree/local_XY

I used a concept similar to this plugin https://github.com/gareth-cross/rviz_satellite

The GPS coordinates can be inserted manually or calculated automatically from NavSatFix message and the frame of the robot where the GPS receiver is located.

Before submitting a PR, it would be nice if someone takes a look at it.

The shortcomins of setting the local_xy_origin in __swri_transform_util/initialize_origin.py__ is that I can hardly find the relationship between the robot frame and a geolocalized fixed frame. This bransh provides a way to do that.

facontidavide, 2017-11-11:
[image]

evenator, 2017-11-13:
Looking over your changes, it looks like `mapviz` *always* publishes a `/local_xy_origin`. Is that the case?

If so, your changes won't work as part of a live robot system that depends on `local_xy_util` for things besides `mapviz`.

facontidavide, 2017-11-13:
If I do that, it is not intentional. I will double check the code.

facontidavide, 2017-11-13:
The local_xy_origin is always advertized but published only here:

https://github.com/Eurecat/mapviz/blob/local_XY/mapviz/src/mapviz.cpp#L1556

I changed that https://github.com/Eurecat/mapviz/commit/9675638b64c418dd5a32a0f0c8ce36525dac2bff

Nevertheless, I wonder if it wouldn't be easier, instead of publishing __local_xy_origin__, to directly Re-Initialized my local instances of __LocalXyWgs84Util__ and __TransformManager__.

This is of course related to the other issue we are discussing https://github.com/swri-robotics/marti_common/pull/499

facontidavide, 2017-11-14:
Ok, I fixed quite a lot of things in my branch here: https://github.com/Eurecat/mapviz/tree/local_XY
No modification of TransfomManager is necessary, even if it would be nice to expose in the public API the swri_trasnform_util::TransformManager::local_xy_util_ as I show here:

https://github.com/facontidavide/marti_common/blob/transform/swri_transform_util/include/swri_transform_util/transform_manager.h#L179

The main differences of the last commit are:

1) All the plugin now do share the instance of TransformManager instantiated by mapviz, instead of creating their own.
https://github.com/Eurecat/mapviz/blob/local_XY/mapviz/include/mapviz/mapviz_plugin.h#L66

2) There allows me to correctly reset everything here https://github.com/Eurecat/mapviz/blob/local_XY/mapviz/src/mapviz.cpp#L1588

After many attempts, this is the only solution I found, since calling Initialize() wasn't sufficient.

__From my side, I start being confident enough to submit a PR...__

malban, 2017-11-14:
Calling initialize at https://github.com/Eurecat/mapviz/blob/local_XY/mapviz/include/mapviz/mapviz_plugin.h#L73 might cause problems for some plug-ins which don't currently expect to have initialize called multiple times.  Only calling the subclass initialize method if 'initialized_' is false would prevent that.

There is also the issue of data that has already been transformed becoming obsolete if the origin changes.  One way to deal with that is to add a virtual resetTransforms function that does nothing by default but that individual plugins could implement to clear obsolete data.

KindDragon, 2019-08-07:
You can use customized `swri_transform_util/initialize_origin.py` from my PR https://github.com/swri-robotics/marti_common/pull/544 that allow to specify `local_xy_custom_topic`