swri-robotics/mapviz #487: Transformation for odometry and multiple objects

fehlfarbe, 2017-05-30:
Hi,

I want to display a car with GPS and other objects that send their positions via VANET messages in rviz and mapviz.
I'm new to ROS and don't have much experience with tf so please correct me if I'm wrong.
The `map` frame is the global position of my car with lat/lon values but no heading because the x-axis should be aligned to east and y-axis to north. So I can add the heading in the `odom` frame that is child of `map`.

So when I start this launchfile:
```
<launch>
    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" output="screen" >
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="auto"/>
    </node>

    <node pkg="mapviz" type="mapviz" name="mapviz"/>
</launch>
```
I have the tree `wsg84 -> map -> odom`

In mapviz I have the fixed frame `map__identity` and the target frame `map` (without heading) or `odom` with heading. With these settings I can see the map tiles and my current car position (NavSatFix). But the view doesn't follow the car (only heading with target frame `odom`) because `map` is set to the first GPS fix.
Shouldn't the `map` frame contain the offset in meters to a `world` frame which is at the first GPS fix:
`wgs84 -> world -> map -> odom` ?
And how do I calculate this `world -> map` offset? 

My second (similar) problem is that I get the lat/lon position of objects via VANET messages. I'm publishing these objects as MarkerArray with lat/lon values for pose position and `map` frame_id (related to my first question this should be `world`?). But the objects only appear at the first GPS fix position.

Wouldn't it be easier to transform all lat/lon coordinates to UTM so the origin is fixed at lat=lon=0° and I only have absolute positions?

malban, 2017-05-30:
To make mapviz follow a target TF frame you will need to publish a TF frame and transform that defines the relationship between the vehicle (usually called 'base_link') and the fixed frame (usually 'map' or 'odom')

The 'map' frame in this context a global coordinate system and is cartesian, represented in meters not lat/lon.  The 'map' frame is tied to the WGS84 lat/lon coordinate system using a projection from the ellipsoid to the 'map' plane.  This projection is defined with the initialize_origin.py node such that 0,0 in the 'map' frame corresponds with the origin lat/lon provided (in the case of 'auto' this is the first GPS message received).  The projection between wgs84 and map cannot be represented in TF since it involves more than a translation and rotation.  It is handled by mapviz (and anything using swri_transform_util) by subscribing to the published local origin as well as TF and performing the necessary projection.

The 'odom' frame usually refers to a local coordinate system where the origin (0,0) is where ever the vehicle started from. The relationship between 'map' and 'odom' is updated such that the vehicle position remains consistent even if the estimates with respect to each coordinate system drift with respect to one another.  That is, at time 0 the position of the vehicle in 'odom' in (0,0) and the position of the vehicle in 'map' might be (35234,18853) and this defines the relationship between the 'odom' frame and 'map'.  As the vehicle moves, if the state estimates of the vehicle in the local odom space and the global space remain consistent, then the transform between odom and map would be constant.  Generally this is not the case though, since the vehicle position with respect to odom generally represents a form of dead-reckoning without global corrections, while the position with respect to map includes global corrections from GPS or SLAM.  Therefore, the relationship between 'odom' and 'map' will tend to drift over time.

The car TF frame would probably be called 'base_link' or 'vehicle'.  This will also be represented in meters and have an origin (0,0) on the center of the rear axel (or somesuch).  This frame is what sensors on the vehicle would be referenced from.  The 'base_link' frame might be related to the 'map' frame directly or be connected to the 'odom' frame depending on how the localization of the vehicle is calculated.  In TF the transform between 'base_link' and 'map' or 'odom' would be represented in meters. 

The map_identity frame is just a kluge to force TF to make the map frame available to mapviz if no other frames have been published.

I'm not sure what your 'world' frame is representing.

For your first issue, I think you need to create a node which subscribes to your GPS vehicle position message and broadcasts the vehicle frame transform as related to map.  This is something that we should maybe add to swri_transform_util, but doesn't currently exist.  It should only be a few lines.  It would consist of creating a swri_transform_util::LocalXyWgs84Util object, subscribing to the GPSFix messages and using the transform util object to convert between 'wgs84' and 'map' points, then broadcasting the 'map' point as the origin of the vehicle frame.  You will also need to set the orientation of the vehicle in the transform, but this will be the same in 'map' as it is in 'wgs84'.  Here is an example of the local xy class being used: https://github.com/swri-robotics/mapviz/blob/172be67e4ebbe3c1c9b3c06e5c2ac1024f627114/mapviz_plugins/src/gps_plugin.cpp#L123-L138   Here is a ROS tutorial on broadcasting transforms: http://wiki.ros.org/tf/Tutorials/Writing%20a%20tf%20broadcaster%20%28C%2B%2B%29

For the second issue, you should just need to set the frame_id of the Markers to '/wgs84' for it to work in mapviz.   Rviz will be unaware of this frame however.  If you want to display these markers in mapviz and rviz you will need to convert the lat/lon values into the map frame (in the same way as the vehicle position described above) and set the frame_id to 'map'.

fehlfarbe, 2017-06-01:
Thanks a lot!
With lat/lon to map frame conversion it's working :)

closed 2017-06-01 by fehlfarbe

uzleo, 2017-06-01:
@fehlfarbe Hi..I also tried to do this lat/lon -> map-frame conversion but couldn't get it working. Could you help me in this regard?

fehlfarbe, 2017-06-01:
That's my snippet:

``` 
#include <swri_transform_util/transform_util.h>

double x, y; // points for map
swri_transform_util::LocalXyFromWgs84(object_boxes.markers[0].pose.position.y, // latitude of point
                                              object_boxes.markers[0].pose.position.x, // longitude of point
                                              pose_origin.pose.position.y, // latitude for localXYorigin
                                              pose_origin.pose.position.x, // longitude localXYorigin
                                              x,
                                              y);
```

Double-check the lat/lon order in the transform function.

uzleo, 2017-06-01:
Hey thanks a lot. This works!

malban, 2017-06-01:
You can also just create a LocalXyWgs84Util object in the node initialization and it will automatically subscribe to the /local_xy_origin message.

```
#include <swri_transform_util/transform_util.h>

LocalXyWgs84UtilPtr local_xy_util;  // member or global variable

...

// Node constructor or initialization function.
local_xy_util = boost::make_shared<LocalXyWgs84Util>();

...
// Message callback

double x, y;
bool success = local_xy_util.ToLocalXy(
  object_boxes.markers[0].pose.position.y,  // latitude
  object_boxes.markers[0].pose.position.x,  // longitude
  x,  // map x
  y) ; // map y

// returns false if no local origin has been received yet

```

Then you can do:
bool success =

malban, 2017-06-01:
You can also treat it more like a TF transform by using the swri_transform_util::TransformManager.
This can be useful if the input data could either be lat/lon or some actual TF frame.

```
#include <swri_transform_util/transform_manger.h>

swri_transform_util::TransformManager tf_manager;  // member or global variable.

...

// Node constructor or initialization function.
tf_manager.Initialize();

...

// Message callback

swri_transform_util::Transform transform;
// Assuming msg.header.frame_id is "/wgs84" if the data is actually lat/lon
if (tf_manager.GetTransform("map", msg.header.frame_id, msg.header.stamp, transform)
{
  tf::Vector3 point(
     object_boxes.markers[0].pose.position.x, 
     object_boxes.markers[0].pose.position.y, 
     object_boxes.markers[0].pose.position.z);
  point = transform * point;
}
```

fehlfarbe, 2017-06-01:
I tried this before but I got an "Couldn't find an AF_INET address for" error so I changed to `swri_transform_util::LocalXyFromWgs84()` function.

malban, 2017-06-01:
If you have a more detailed error message I could try and figure out whats going on.   I haven't seen this before related to the transform manager.   It is important to call the initialization function after ros::init()

Maybe it is this?  http://answers.ros.org/question/163556/how-to-solve-couldnt-find-an-af_inet-address-for-problem/

fehlfarbe, 2017-06-01:
That's the whole error message:
`Couldn't find an AF_INET address for []`

> It is important to call the initialization function after ros::init()

That was the problem! I initialized the LocalXyWgs84Util before ros::init(). Now it's working, thanks :)

malban, 2017-06-01:
Great :)  No problem.

kylerlaird, 2018-03-10:
My INS is publishing a transform (/ins) which uses UTM coordinates for a zone which is specified as a parameter.

Is there a way to use this transform directly with mapviz?  Do I need to convert it back to wgs84 first?

[2018-03-10-08-33-31.bag.zip](https://github.com/swri-robotics/mapviz/files/1799658/2018-03-10-08-33-31.bag.zip)

malban, 2018-03-12:
I think this can work.  

Here is a launch file which worked for me with your bag file.  I had to add an additional static transform between 'world' and the local origin 'map'.  This could be automated if you implement your own the initialize_origin.py script to calculate and publish that static transform for whatever the local origin is defined as.  We usually set the local origin to 'auto', which has the initializer subscribe to a GPS message and use the first valid lat/lon it gets.

```
<launch>
  <!-- use sim time when playing back bag files with the clock flag -->
  <param name="use_sim_time" value="true"/>

  <node pkg="mapviz" type="mapviz" name="$(anon mapviz)" required="true">
  </node>

  <!-- The local origin needs to be in the general location of where the data is. -->
  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="map"/>
    <param name="local_xy_origin" value="custom"/>
    <rosparam param="local_xy_origins">
      [{ name: custom,
         latitude: 40.887893,
         longitude: -87.193331,
         altitude: 169.0,
         heading: 0.0}]
    </rosparam>
  </node>

  <!-- The 'world' frame published by the INS is 0, 0 of the UTM zone.  We want to relate
       that to the local origin we defined above.  This can be done using the UTM 
       coordinates for the of the local origin.  Now TF knows how to go to/from 
       'map' <-> 'world' <-> 'ins' and maviz knows how to go between TF and lat/lon and 
       UTM through the local origin of 'map'. -->
  <node pkg="tf" 
        type="static_transform_publisher" 
        name="world_map_transform" 
        args="483713.04 4526330.23 0 0 0 0 world map 100"  />

  <!-- Needed only for searching for services -->
  <node pkg="rosapi" type="rosapi_node" name="rosapi"/>

</launch>
```
These are the transforms that exist with this launch file.

[image]

kylerlaird, 2018-03-15:
Thank you for your quick response, Marc.  I was excited to try your suggestion but I still get this:
```
[ERROR] [1521146572.345070001]: Error: No transform between  and ins
[ERROR] [1521146572.465333052]: Error: No transform between /wgs84 and ins
[ERROR] [1521146573.459788691]: Error: No transform between map and ins
```

Is there some other configuration I need to do?

malban, 2018-03-16:
Using the launch file above and this mapviz config file:

```
capture_directory: ~
fixed_frame: map
target_frame: ins
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1241
window_height: 747
view_scale: 0.1558985
offset_x: 9.665708
offset_y: -17.14884
use_latest_transforms: true
background: "#a0a0a4"
displays:
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: <--YOUR BING API KEY GOES HERE-->
      source: Bing Maps (terrain)
  - type: mapviz_plugins/tf_frame
    name: new display
    config:
      visible: true
      collapsed: false
      frame: ins
      color: "#00ff00"
      draw_style: arrows
      position_tolerance: 0
      buffer_size: 0
      static_arrow_sizes: false
      arrow_size: 25
```
I get the following result when running `rosbag play  2018-03-10-08-33-31.bag --clock`

[image]

Things that you may want to check.
- are you playing the bag with --clock and does the launch file set use_sim_time?
- is the fixed frame 'map'?
- is the target frame 'ins'?
- what data is trying to be transformed with an empty frame id?
  - `[ERROR] [1521146572.345070001]: Error: No transform between  and ins`
- with the above launch file, what is the result of `rosrun tf view_frames` when playing `rosbag play  2018-03-10-08-33-31.bag --clock`