swri-robotics/mapviz #500: tile map failing to build on kinetic buildfarm

tfoote, 2017-08-10:
One example: 
http://build.ros.org/view/Kbin_uX64/job/Kbin_uX64__tile_map__ubuntu_xenial_amd64__binary/49/console

```
/tmp/binarydeb/ros-kinetic-tile-map-0.2.3/src/tile_map_plugin.cpp
00:08:16.706 /tmp/binarydeb/ros-kinetic-tile-map-0.2.3/src/tile_map_plugin.cpp: In member function ‘virtual void tile_map::TileMapPlugin::SaveConfig(YAML::Emitter&, const string&)’:
00:08:16.706 /tmp/binarydeb/ros-kinetic-tile-map-0.2.3/src/tile_map_plugin.cpp:408:31: error: ‘trim_copy’ is not a member of ‘boost’
00:08:16.706                 YAML::Value << boost::trim_copy(bing_source->GetApiKey().toStdString());
00:08:16.706                                ^
00:08:16.706 /tmp/binarydeb/ros-kinetic-tile-map-0.2.3/src/tile_map_plugin.cpp:411:31: error: ‘trim_copy’ is not a member of ‘boost’
00:08:16.706                 YAML::Value << boost::trim_copy(ui_.source_combo->currentText().toStdString());
00:08:16.706                                ^
00:08:16.716 CMakeFiles/tile_map_plugin.dir/build.make:73: recipe for target 'CMakeFiles/tile_map_plugin.dir/src/tile_map_plugin.cpp.o' failed
00:08:16.716 make[4]: *** [CMakeFiles/tile_map_plugin.dir/src/tile_map_plugin.cpp.o] Error 1
00:08:16.716 make[3]: *** [CMakeFiles/tile_map_plugin.dir/all] Error 2****
```

It has been fixed in #496 but not released.

tfoote, 2017-08-10:
This is also an issue in the latest indigo release: http://build.ros.org/view/Ibin_uT64/job/Ibin_uT64__tile_map__ubuntu_trusty_amd64__binary/

pjreed, 2017-08-11:
Should be fixed by ros/rosdistro#15765 and ros/rosdistro#15766.

closed 2017-08-11 by pjreed