swri-robotics/mapviz #501: multires_image with transparent regions

ramboto, 2017-08-11:
Hello,
When I try to overlay an image on a tile map in mapviz, I notice the transparent regions of the image get colored black(I am using mapproxy with google map layer). If I put the same image on the google-earth application (with kml file), the transparent region is displayed correctly(I see the google map layer). Are there any ideas on how I can resolve this?

pjreed, 2017-08-11:
Are you drawing the image with Mapviz's `image` plugin?  I suspect that alpha blending may not be enabled when it's drawing the image; we probably haven't tested it with images that have alpha channels.  I'll do that sometime when I have the time for it, unless somebody else gets there first...

ramboto, 2017-08-11:
Ah sorry I think I wasn't specific in my description. I'm not using the Image plugin(sensor_msgs::Image), but the Multi-res Image plugin. I created a geo file giving the path to my image that I want to overlay onto a tile_map plugin. When I run mapviz and choose the multi-res image plugin, I see the image being overlaid correctly onto the tile_map plugin, but the transparent regions of my multi-res image turn out black.

If I use the same image on google earth the transparency is displayed correctly(or I can see the map layer underneath where the image is transparent)

pjreed, 2017-08-14:
Ok, I took a look at this and was able to reproduce it, and I think it's because the `multires_image` plugin is internally storing GL textures without an alpha channel.  Should be easy to fix.

closed 2017-08-22 by pjreed

omirbod, 2017-08-22:
Great thank you.

PR #504 (2017-08-22): Support transparent tiles in multires_image (Kinetic)
The multires_image plugin was internally loading textures without an
alpha channel, so tiles that had transparent regions would show
black instead.  This will cause transparent regions to render properly.

Fixes #501
commit e30adc7d (2017-08-22): Support transparent tiles in multires_image

The multires_image plugin was internally loading textures without an
alpha channel, so tiles that had transparent regions would show
black instead.  This will cause transparent regions to render properly.

Fixes #501
PR #503 (2017-08-22): Support transparent tiles in multires_image (Jade)
The multires_image plugin was internally loading textures without an
alpha channel, so tiles that had transparent regions would show
black instead.  This will cause transparent regions to render properly.

Fixes #501
commit 96969189 (2017-08-22): Support transparent tiles in multires_image

The multires_image plugin was internally loading textures without an
alpha channel, so tiles that had transparent regions would show
black instead.  This will cause transparent regions to render properly.

Fixes #501