swri-robotics/mapviz #507: Example bag file

rocklinsuv, 2017-09-06:
Dear mapviz developer,

I'm new to mapviz and unfortunately I didn't find any relevant tutorials online except for this github page. Is it possible for you to provide me an example bag file so that I can see what kind of messages/topics should I publish/subscribe to? Any help would be very much appreciated. 

Best Regards,
rocklinsuv

luiscarlosfeliz, 2018-07-17:
Did you manage to do something with this? I'm having the same problem. I have gps data but cant even load a terrain map on mapviz..

tuandle, 2018-07-18:
Hi @rocklinsuv , could you kindly give a walkthrough from the very beginning to successfully load the map and draw gps position on it? I am really confused when reading through all the suggested issues. 
When I launch my mapviz launch file and play my rosbag, all I got is
`[139665220650752] [/initialize_origin/initialize_origin:105]: Local XY origin is "auto"
[139665220650752] [/initialize_origin/initialize_origin:108]: Local XY frame ID is "/map"
[139665220650752] [/initialize_origin/initialize_origin:125]: Subscribed to NavSat on /fix
[139664216291072] [/initialize_origin/gps_callback:58]: Got NavSat message. Setting origin and unsubscribing from NavSat`
And I have no clue what to do next.
Thank you for your help!

rocklinsuv, 2018-07-19:
@tuandle 

Previously I worte a tutorial about how to use MapViz in ROS environment. Due to some reasons I can't share the entire document to you, but here are some screenshots which may help you finish what you need to do:

[image]
[image]
[image]

For asctec_msgs/GPSdata.msg please refer to this link: http://docs.ros.org/diamondback/api/asctec_msgs/html/msg/GPSData.html
What you need is write your own publisher, prepare a "NavSatFix.msg" and publish to /navsat/fix.

BR,
rocklinsuv

tuandle, 2018-07-19:
@rocklinsuv 
Thank you for your fast reply.
I did follow everything you showed yesterday. I install the map proxy and was sure that it is up and running. However, the status after adding tile_map is "no tranform between /wgs84" and it gets stuck there, no matter what I do. Whenever I add a new element (e.g gps), mapviz crashes. 
I wonder if you encountered these problems before?

rocklinsuv, 2018-07-19:
@tuandle 
[image]

Something like this? I haven't used MapViz for nearly half a year, and this error appeared. However I currently don't have access to GoogleMap. I remembered that the default map of MapViz is OpenStreetMap. Can your MapViz use this map source?

tuandle, 2018-07-19:
@rocklinsuv 
Yes, that is the error I have. I have no problem access to google map with mapproxy by danielsnider. 
I am not sure how to make Mapviz use OpenStreetMap source. Do you know where I should look for it? 
And do you suspect the error happens because MapViz can't access to some map service to download map images?

rocklinsuv, 2018-07-19:
@tuandle 

After you execute the command "mapproxy‐util serve‐develop ‐b 127.0.0.1:8080 mapproxy_online.yaml", try to use a browser and type in "127.0.0.1:8080" in the address bar, can you run the MapProxy demo?
Switching map sources is done via editing the "mapproxy_online.yaml", please refer to mapproxy tutorial :D

tuandle, 2018-07-19:
I did follow daniel's instruction to run mapproxy. I didn't run your command since there is no mapproxy_online.yaml at [danielsnider's repository](https://github.com/danielsnider/MapViz-Tile-Map-Google-Maps-Satellite). I was able to run map proxy demo by following daniel's instruction.
[image]
Did you create the file mapproxy_online.yaml by yourself?

malban, 2018-07-19:
@tuandle If you run `roslaunch mapviz mapviz.launch` and add the tile_map plug-in (it should default to the 'Stamen (terrain)' source) are you able to see the map?

Running `roslaunch mapviz mapviz.launch` should look like this.  If not, you can remove ~/.mapviz_config to clear the configuration.
[image]

Adding the tile_map should look like this:
[image]

At this point you can also add a 'navsat' plugin to display a GPS topic on the map.  You can also display any other geo-referenced data published in the '/wgs84' or '/utm' frames, where '/wgs84' is x:longitude, y:latitude and '/utm' is x:easting, y:northing of the UTM zone the local origin is defined in.

You can also display any data in a normal TF frame as long as it is correctly connected to the local origin frame defined with initialize_origin.  In this case '/far_field'

The local origin should ideally be defined somewhere in the vicinity of the data being displayed.  In this case, the origin is in San Antonio, TX.  Using the 'auto' option for initialize origin will use the first GPS message it receives to set the origin.  This can be convenient, but I would recommend setting a static one initially to simplify things.

It is also possible to a create custom tile server and then configure the 'tile_map' plugin to point to that.  I would recommend confirming things work on the default map first though.

tuandle, 2018-07-19:
@malban 
Thank you for your input.
I run the example launch file:
`<?xml version="1.0"?>
<launch>
  <env name="ROSCONSOLE_FORMAT" value="[${thread}] [${node}/${function}:${line}]: ${message}"/>
  <arg name="print_profile_data" default="false"/>
  <node pkg="mapviz" type="mapviz" name="$(anon mapviz)" required="true">
    <param name="print_profile_data" value="$(arg print_profile_data)"/>
  </node>
  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/far_field"/>
    <param name="local_xy_origin" value="swri"/>
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 29.45196669,
         longitude: -98.61370577,
         altitude: 233.719,
         heading: 0.0},         
       { name: back_40,
         latitude: 29.447507,
         longitude: -98.629367,
         altitude: 200.0,
         heading: 0.0}]
    </rosparam>
  </node>
  <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /far_field /origin 100"  />
  
  <node pkg="rosapi" type="rosapi_node" name="rosapi"/>
</launch>`
After adding the tile_map, I got this:
[image]

The status always shows that I miss the transform from /wsg84 frame.
And I also notice except for tile_map, I can't add anything else. It always shows I have error 245.
What else should I do? 
Thank you for your help!
P.S. I use kinetic branch, but I guess it is not a problem, right?

malban, 2018-07-19:
When you move the mouse cursor over the canvas, are the latitude and longitude displayed on the bottom panel like this?
[image]

or are they missing like this?
[image]

tuandle, 2018-07-19:
@malban 
Yes, the latitude and longitude changes when I move the cursor.
[image]

malban, 2018-07-19:
If that's the case, then the tile_map plug-in should have no issue transforming the data.  It is expected for the plug-in to print this error when it is first created, since it needs to subscribe to and receive the local origin before it can transform the data.

It does look suspicious that the error appears to be 'No transform between /wgs84 and  '  instead of 'No transform between /wgs84  and /far_field.'

If I change the fixed frame to /far_fieldx is get this:
[image]

Do you see the same message? and when you set it back to '/far_field'?

Are you using the debian binaries or building from source?

tuandle, 2018-07-19:
Yes, the message is the same: Error: No transform between /wgs84 and when I change the fixed frame to /far_filedx and back. The message stays the same.
I build from source.
[image]

malban, 2018-07-19:
Can you try using the debian package instead and seeing if you get the same behavior?

rocklinsuv, 2018-07-19:
@malban @tuandle One thing for you guys to consider. I installed MapProxy from source (following the tutorial) and MapViz via git. I remembered you said using Docker for MapProxy installation. Have you tried to install it without Docker? Though it is a bit painful. The reason why I abanadoned Docker was the version confliction with my Ubuntu.

tuandle, 2018-07-19:
@malban 
I purged mapviz package and installed from source. However, the error is still the same.
[image]
[image]
@rocklinsuv I removed Docker to make sure that I don't have anything suspicious problem, yet it didn't help either...

tuandle, 2018-07-19:
@malban I am sorry that I mistyped. I purged everything related to mapviz from my workspace.
I instal from debian by using:
`sudo apt-get install ros-$ROS_DISTRO-mapviz ros-$ROS_DISTRO-mapviz-plugins ros-$ROS_DISTRO-tile-map ros-$ROS_DISTRO-multires-image`
And I repeated all the tests you mentioned: add tile_map, move cursor to see lat/lon changes, change fix_frame. Yet the error is the same.

malban, 2018-07-19:
Hmm.  I'm not sure what to look at from here other than stepping through the code.  

That error should be triggered here:
https://github.com/swri-robotics/mapviz/blob/kinetic-devel/tile_map/src/tile_map_plugin.cpp#L328

You could try and trace why the target_frame_ variable is empty when it should contain the value of the Fixed Frame drop down.  It sounds like a bug, but I am unable to reproduce it.

tuandle, 2018-07-19:
@malban I will try to trace the error in the tile_map plugin code and get back. Thank you so much for your help!

malban, 2018-07-19:
TileMapPlugin::Transform() function is called here:
https://github.com/swri-robotics/mapviz/blob/kinetic-devel/mapviz/include/mapviz/mapviz_plugin.h#L122

and the target_frame_ variable should be set here:
https://github.com/swri-robotics/mapviz/blob/kinetic-devel/mapviz/include/mapviz/mapviz_plugin.h#L150

Frank1Ding, 2019-09-11:
@malban 
Sorry for bothering. I’m new to mapviz and I just display the tile map and GPS location successfully. But I want use Mapviz for navigation on the road, just like google map etc. But I have no idea about this. I’m wondering if you can give me some advice.

Frank1Ding, 2019-09-11:
For example，is it possible to use Mapviz to generate a route on the road from the current position to goal position base on the online or offline map.

Frank1Ding, 2019-09-11:
> > Sorry for bothering. I’m new to mapviz and I just display the tile map and GPS location successfully. But I want use Mapviz for navigation on the road, just like google map etc. But I have no idea about this. I’m wondering if you can give me some advice.
> 
> @Frank1Ding You'll need to be a bit more specific on what your issue is or what advice you need.

For example，is it possible to use Mapviz to generate a route on the road from the current position to goal position base on the online or offline map.

malban, 2019-09-11:
Mapviz is really just a visualization tool for ROS like http://wiki.ros.org/rviz.

There are some plug-ins that can help with routing like:
 - move_base: uses http://wiki.ros.org/move_base
 - plan_route: uses [marti_nav_msgs::PlanRoute
](https://github.com/swri-robotics/marti_messages/blob/master/marti_nav_msgs/srv/PlanRoute.srv) service

In both cases, the plug-in is just calling services provided by what ever underlying navigation system has been implemented with ROS.

Frank1Ding, 2019-09-11:
That is very helpful. Thanks very much for your help.

ZahidHasan107, 2021-08-08:
Dear mapviz developer,
Kindly help me out. How can I solve this problem? 
[image]

matt-attack, 2021-08-12:
@ZahidHasan107 it appears you do not have an origin set. You can set one by adding something like below to your launch:

`  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/far_field"/>
    <param name="local_xy_origin" value="swri"/>
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 29.45196669,
         longitude: -98.61370577,
         altitude: 233.719,
         heading: 0.0},         
       { name: back_40,
         latitude: 29.447507,
         longitude: -98.629367,
         altitude: 200.0,
         heading: 0.0}]
    </rosparam>
  </node>`

Make sure to set your desired lat/long. You can also have the node set the origin automatically upon receipt of a GPS message which I can also show an example of if that would be helpful.

closed 2021-09-02 by danthony06