swri-robotics/mapviz #562: Can't roslaunch

iii2, 2018-01-03:
TL:DR - roslaunch doesn't able to find libopencv_imgcodecs3.so.3.3 

Hi there, I'm having trouble to launch the application. I have ros lunar (but also did some tests with kinetic - no help), I installed libopencv-dev, and by default opencv3 inside - /opt/ros/lunar/share/opencv3.
I sourced:
source /opt/ros/lunar/setup.bash
source ~/catkin_ws/devel/setup.bash

LD_LIBRARY_PATH:
/home/pitay/catkin_ws/devel/lib:/opt/ros/lunar/lib:/usr/local/lib:/opt/ros/lunar/share

and tried to launch without success:
roslaunch mapviz mapviz.launch 

I'm quite noob and would appreciate very much any help.
I did a full restore to my linux environment today, so it's all almost stock system.

The error:
ROS_MASTER_URI=http://localhost:11311

process[mapviz_Itay_PC_5733_8973223146860273745-1]: started with pid [5750]
process[initialize_origin-2]: started with pid [5751]
process[swri_transform-3]: started with pid [5752]
process[rosapi-4]: started with pid [5753]
/home/pitay/catkin_ws/devel/lib/mapviz/mapviz: error while loading shared libraries: libopencv_imgcodecs3.so.3.3: cannot enable executable stack as shared object requires: Invalid argument
================================================================================REQUIRED process [mapviz_Itay_PC_5733_8973223146860273745-1] has died!
process has died [pid 5750, exit code 127, cmd /home/pitay/catkin_ws/devel/lib/mapviz/mapviz __name:=mapviz_Itay_PC_5733_8973223146860273745 __log:=/home/pitay/.ros/log/1d8aae36-f0ad-11e7-92bf-74d435f5780f/mapviz_Itay_PC_5733_8973223146860273745-1.log].
log file: /home/pitay/.ros/log/1d8aae36-f0ad-11e7-92bf-74d435f5780f/mapviz_Itay_PC_5733_8973223146860273745-1*.log
Initiating shutdown!
================================================================================
[rosapi-4] killing on exit
[swri_transform-3] killing on exit
[initialize_origin-2] killing on exit
[mapviz_Itay_PC_5733_8973223146860273745-1] killing on exit
Traceback (most recent call last):
  File "/opt/ros/lunar/lib/swri_transform_util/initialize_origin.py", line 31, in <module>
    from gps_common.msg import GPSFix
  File "/opt/ros/lunar/lib/python2.7/dist-packages/gps_common/msg/__init__.py", line 1, in <module>
    from ._GPSFix import *
  File "/opt/ros/lunar/lib/python2.7/dist-packages/gps_common/msg/_GPSFix.py", line 5, in <module>
    import genpy
  File "/opt/ros/lunar/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/lunar/lib/python2.7/dist-packages/genpy/message.py", line 46, in <module>
    import genmsg
  File "/opt/ros/lunar/lib/python2.7/dist-packages/genmsg/__init__.py", line 33, in <module>
    from . base import MSG_DIR, SRV_DIR, EXT_MSG, EXT_SRV, SEP, log, plog, InvalidMsgSpec, log_verbose, MsgGenerationException
  File "/opt/ros/lunar/lib/python2.7/dist-packages/genmsg/base.py", line 52, in <module>
    import inspect, pprint
  File "/usr/lib/python2.7/pprint.py", line 35, in <module>
    """
KeyboardInterrupt
Traceback (most recent call last):
  File "/opt/ros/lunar/lib/rosapi/rosapi_node", line 34, in <module>
    import rospy
  File "/opt/ros/lunar/lib/python2.7/dist-packages/rospy/__init__.py", line 47, in <module>
    from std_msgs.msg import Header
  File "/opt/ros/lunar/lib/python2.7/dist-packages/std_msgs/msg/__init__.py", line 1, in <module>
    from ._Bool import *
  File "/opt/ros/lunar/lib/python2.7/dist-packages/std_msgs/msg/_Bool.py", line 5, in <module>
    import genpy
  File "/opt/ros/lunar/lib/python2.7/dist-packages/genpy/__init__.py", line 34, in <module>
    from . message import Message, SerializationError, DeserializationError, MessageException, struct_I
  File "/opt/ros/lunar/lib/python2.7/dist-packages/genpy/message.py", line 44, in <module>
    import yaml
  File "/usr/lib/python2.7/dist-packages/yaml/__init__.py", line 8, in <module>
    from loader import *
  File "/usr/lib/python2.7/dist-packages/yaml/loader.py", line 4, in <module>
    from reader import *
  File "/usr/lib/python2.7/dist-packages/yaml/reader.py", line 47, in <module>
    class Reader(object):
  File "/usr/lib/python2.7/dist-packages/yaml/reader.py", line 140, in Reader
    NON_PRINTABLE = re.compile(u'[^\x09\x0A\x0D\x20-\x7E\x85\xA0-\uD7FF\uE000-\uFFFD\U00010000-\U0010ffff]')
  File "/usr/lib/python2.7/re.py", line 194, in compile
    return _compile(pattern, flags)
  File "/usr/lib/python2.7/re.py", line 249, in _compile
    p = sre_compile.compile(pattern, flags)
  File "/usr/lib/python2.7/sre_compile.py", line 576, in compile
    code = _code(p, flags)
  File "/usr/lib/python2.7/sre_compile.py", line 558, in _code
    _compile_info(code, p, flags)
  File "/usr/lib/python2.7/sre_compile.py", line 536, in _compile_info
    _compile_charset(charset, flags, code)
  File "/usr/lib/python2.7/sre_compile.py", line 232, in _compile_charset
    flags & SRE_FLAG_UNICODE):
  File "/usr/lib/python2.7/sre_compile.py", line 274, in _optimize_charset
    r = range(av[0], av[1]+1)
KeyboardInterrupt
shutting down processing monitor...
... shutting down processing monitor complete

daniel-dsouza, 2019-09-16:
At this point @iii2, I would say to either revert back to ROS Kinetic or upgrade to Melodic. Lunar is no longer a supported distribution.

After sourcing the appropriate ROS environment `setup.bash`, have you tried `rosrun mapviz mapviz`?

pjreed, 2019-09-16:
I don't know what version of Linux you're using, but note that on Ubuntu 16.04, the package `libopencv-dev` provides OpenCV 2.4.9.  ROS Lunar (and mapviz) uses OpenCV 3.3, which is provided by the package `ros-lunar-opencv3`.  If you use `rosdep` to install your dependencies, or if you use apt to install the `ros-lunar-mapviz` package, it should automatically install that.

If OpenCV3 is properly installed, you should have libraries at these locations:
```bash
# ll /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so*
lrwxrwxrwx 1 root root      27 Jun  3 08:59 /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so -> libopencv_imgcodecs3.so.3.3
lrwxrwxrwx 1 root root      29 Jun  3 08:59 /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3 -> libopencv_imgcodecs3.so.3.3.1
-rw-r--r-- 1 root root 4277928 Jun  3 08:59 /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3.1
```

In addition, if you run `ldd` on your mapviz binary, if everything is linked properly, it should point to opencv libraries like so:
```bash
# ldd /opt/ros/lunar/lib/mapviz/mapviz |grep opencv
        libopencv_videoio3.so.3.3 => /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_videoio3.so.3.3 (0x00007f183d760000)
        libopencv_core3.so.3.3 => /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_core3.so.3.3 (0x00007f183c8dd000)
        libopencv_imgproc3.so.3.3 => /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_imgproc3.so.3.3 (0x00007f183a0a7000)
        libopencv_imgcodecs3.so.3.3 => /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_imgcodecs3.so.3.3 (0x00007f1839a92000)
        libopencv_video3.so.3.3 => /opt/ros/lunar/lib/x86_64-linux-gnu/libopencv_video3.so.3.3 (0x00007f1831ac0000)
```

If you're sure OpenCV 3 is installed, the first thing I'd try doing is removing `libopencv-dev` (and any other traces of OpenCV 2).  ROS doesn't need it, and if you have any other ROS packages that are using OpenCV, they may be accidentally including OpenCV 2 in the build environment and inadvertently breaking other packages.

danthony06, 2021-06-11:
I'm closing this because it's referring to a non-support version of ROS, and we have not been able to reproduce this on the current versions.

closed 2021-06-11 by danthony06