swri-robotics/mapviz #567: Give robot_image a gps coordinate for its frame

cweeks12, 2018-01-25:
What frame do you use/have to create to put a robot_image at a certain GPS coordinate within mapviz? 

What I'm trying right now is I've written a simple transform to echo point_click events to the `rover` frame and having the robot_image just go to where I clicked. It works when I use `origin` as the parent frame within the transform and breaks when I refer to `/wgs84`, but for me to use it to follow the robot when it's actually travelling, I would have to convert the GPS coordinate off of it to something within the local xy space, which I'm unsure on how to do.

I guess what my question is, is how do I set up mapviz to show the rover image given a GPS coordinate. Do I have to write a transform from `/wgs84` to some GPS coordinates or use a dummy transform to get it to work?

pjreed, 2018-01-25:
mapviz uses the `swri_transform_util::TransformManager` from the [marti_common](https://github.com/swri-robotics/marti_common) repository in order to handle transforming points in the `/wgs84` frame.  The easiest way to initialize that is to use a `swri_transform_util/initialize_origin.py`, which takes in a GPSFix or NavSatFix message and then publishes a PoseStamped message to `/local_xy_origin`.  There's some more detailed discussion and an example of how to do so here: https://github.com/swri-robotics/mapviz/issues/256

cweeks12, 2018-01-26:
Okay, so I think I understand all of that. I have a `/local_xy_origin` being published, my map is appearing, and I can make the `robot_image` appear in places within the `origin` frame. I just don't know how to do those conversions on the fly when I get a GPS coordinate back from the robot. 

How do I, either by hand, or by publishing GPS coordinates to a topic, get a conversion from the GPS coordinates to a reference in the local frame, because the `robot_image` plugin only takes a tf frame as its location?

Sorry if this is a simple question and I just don't understand.

malban, 2018-01-26:
Right.  The robot image plugin is expecting to place the image at the origin of a TF frame.  The convention in ROS is to use TF to manage the robot transforms: sensors to robot, robot to map, etc.  So there should be a frame in TF that corresponds to the robot.  In ROS this if often 'base_link'.  That is the frame you would select for the robot_image plugin.

Some part of the system will need to localize the robot with respect to the coordinate systems you care about (global and/or odom).  This could either be something like: http://wiki.ros.org/robot_pose_ekf or it could simply be based on the raw GPS data.  If your wanting to track the vehicle with global coordinates, the data from the GPS sensor will need to be converted to a Cartesian frame using some projection  to work with TF and the rest of the ROS.

swri_transform_util provides tools for defining the transform (initialize_origin.py), and converting between Lat/Lon and TF (TransformManager and LocalXyWgs84Util).

For the robot_pose_ekf case,  a node would need to be created to convert a NavSatFix message to an Odometry message in a TF frame (probably 'map' or 'odom').  The odometry message could be passed to robot_pose_ekf which will generate the TF transforms between 'base_link' and the frame of the Odometry message.  If you want to use the raw GPS data, then the conversion node could also publish the tf transforms.

There are a few things we could probably do to make this easier. 
1. We could add a simple node in swri_transform_util to take a NavSatFix message and output Odometry/TF
2. We could modify the robot image plugin to also be able to track NavSatFix or Odometry messages instead of just TF.

malban, 2018-01-26:
@cweeks12 This pull request may help solve your problem.  It's untested at this point.  You can also use it as an example for how to perform the conversion in your own code.

https://github.com/swri-robotics/marti_common/pull/509

cweeks12, 2018-01-29:
That looks like exactly what I need, and would be a great thing in general for mapviz. A node that acts as a "tf" between gps and the local_xy frame would be awesome. That looks like what the pull request is doing, is that right?

cweeks12, 2018-01-29:
Looking at it, the code in the pull request gets the rotation from the `track` of the NavSatFix message, which says that it's inaccurate at low speeds. Could there be a way to convert just the GPS coordinates to the translation part of the transform, then allow a rotation to be defined from an IMU or something instead of the gps `track`?

pjreed, 2018-01-29:
Once you're at the point of wanting to combine together your GPS fix and IMU measurements to produce more accurate odometry, you're going to want to look into running a state estimator.  I'd recommend taking a look at the [robot_localization](http://docs.ros.org/lunar/api/robot_localization/html/index.html) package to see if that will do what you want.

cweeks12, 2018-01-30:
Alright, I will look into that. I noticed you merged in that other pull request. Thanks for doing that. I'll wait for that to get into a release, then use that to perform the transforms. Y'all rock.

closed 2018-01-31 by cweeks12