swri-robotics/mapviz #574: Needs to be a way to clear data

daveconger, 2018-03-20:
My mapviz keep getting filled up with clutter from successive bag runs.  There needs to be a way to clear data (or re-init plugins).  Restarting mapviz each time is not sufficient since geo maps are not cached and need to redownload each run.

HoomanBean, 2018-03-20:
Yeah the tubes keep getting clogged. Also I need to borrow Jerry's scissors.

malban, 2018-03-20:
Seems like a useful feature.  Base plugin could be modified with a virtual clear data method and individual plugins could be modified to clear their buffers.

What specifically is getting cluttered up?  Things like the odometry track?

malban, 2018-03-20:
Also  @HoomanBean  doesn't seem like a real person, I wouldn't trust anything they say.

daveconger, 2018-03-20:
Off hand, at least the gps display.

malban, 2018-03-21:
@daveconger See if these branches work:

https://github.com/malban/mapviz/tree/clear-history-indigo
https://github.com/malban/mapviz/tree/clear-history-jade

It adds a 'Clear History' button to the menu bar and clears the history of GPS, tf, and odometry tracks, point clouds, laser scans, and markers.  You will need to recompile both mapviz and mapviz_plugins.

HoomanBean, 2018-03-21:
cool beans