swri-robotics/mapviz #575: tile_map not showing

FranziskaHenke, 2018-03-21:
Hi!

I've been using MapViz for ROS indigo for a while now, but for some reason the tile_map plugin never seemed to work for me. If I add the tile_map display it says "OK" but no tiles are displayed with any of the options. I also tried adding custom tile_map sources as well as installing from the ROS apt repository and from source. Everything else is working fine!
Here is my launch-file:

```
<launch>
<node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /base_link 100"  />

<node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="auto"/>
    <rosparam param="local_xy_origins">
        [{ name: Bar,
         latitude: 37.9879772,
         longitude: 23.9078602,
         altitude: 129.69,
         heading: 0.0}]
    </rosparam>
</node>

<node pkg="mapviz" type="mapviz" name="mapviz"/>
</launch>
```
Is my transformation wrong or can this have something to do with my company's proxy settings and how can I test this?
Thank you for your help!

malban, 2018-03-21:
Are you able to add and see the tile_map plugin when you run roslaunch mapviz mapviz.launch?

It looks like maybe you have an issue with the local origin in the launch file you provided.  You have it set to 'auto', which will only publish an origin when it receives a GPS message.  If the initialize_origin.py node doesn't receive a GPS message it will never initialize.  If you set it to 'Bar' instead it should publish immediately.

FranziskaHenke, 2018-03-23:
Hi! Thanks a lot for your quick response.

When I run roslaunch mapviz mapviz.launch I can see the transformation of the GPS message (GPS display ) and also the /local_xy_origin seems to be published when I run rostopic echo /local_xy_origin. Setting the local origin to Bar unfortunately does not make a difference.

I can still add but not see the tile_map plugin. The status says OK.

malban, 2018-03-23:
It's possible there is an issue with your internet connection.  If you run the mapviz.launch file in the mapviz package and add a tile_map plugin you should get something similar to this:
[image]

The config file contains this:
```
capture_directory: ~
fixed_frame: far_field
target_frame: <none>
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1000
window_height: 804
view_scale: 1
offset_x: 0
offset_y: 0
use_latest_transforms: true
background: "#a0a0a4"
displays:
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: ""
      source: Stamen (terrain)
```

You could try loading this exact configuration after running roslaunch mapviz mapviz.launch

If it is a network issue I would expect it to spew `NETWORK ERROR` ROS_ERROR messages in the console.

You could also try using another source like Bing Maps.  With Bing you need to provide an API key.  It looks like you need to restart mapviz after entering in and saving the key.

It is also possible to load tiles locally to bypass the network altogether.  You would select custom source and set the base url to something like:  `file:///home/malban/map_tiles/{level}/{x}/{y}.png`

If none of that works, I'm not sure.  You may need to dig into the plug-in and add some debugging log messages to find out what is going on.

FranziskaHenke, 2018-03-26:
Hi!

My config file looks very similar to your, but still no tile_map and no error message.

I finally got it to work by loading the tiles locally and selecting custom source (using this tool to download the tiles https://github.com/sjev/mapCreator in case somebody else needs it).
Thanks a lot for your support!

Combinacijus, 2020-08-25:
Another solution to use this: [https://github.com/danielsnider/MapViz-Tile-Map-Google-Maps-Satellite](https://github.com/danielsnider/MapViz-Tile-Map-Google-Maps-Satellite)

jkoubs, 2024-07-17:
Hello! Does anyone knows what is the config file synthax the set a Custom WMTS Source?

Basically I want to replicate the following file to use CustomWMTS Source:

`capture_directory: ~
fixed_frame: far_field
target_frame: <none>
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1000
window_height: 804
view_scale: 1
offset_x: 0
offset_y: 0
use_latest_transforms: true
background: "#a0a0a4"
displays:
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: ""
      source: Stamen (terrain)`