swri-robotics/mapviz #580: mapviz failing to build on armhf

mikaelarguedas, 2018-04-18:
Hi @pjreed and mapviz maintainers.

`mapviz` has been released in Lunar but failed to build ever since: http://build.ros.org/view/Lbin_uxhf_uXhf/job/Lbin_uxhf_uXhf__mapviz__ubuntu_xenial_armhf__binary/

Is this something that is expected?
Is `mapviz` planning on supporting armhf platforms ?

If this is not expected to build in the foreseeable future, can you please reply here and I'll make sure the buildfarm doesn;t try to build this package anymore.

Thanks!

pjreed, 2018-04-18:
I've spent a while looking into it before, and making mapviz build on armhf would take a considerable amount of work, as it's using some OpenGL 2 APIs that don't seem to exist on armhf.

None of us actually need to use it on arm, so it probably won't get fixed any time soon; feel freel to disable the build.

mikaelarguedas, 2018-04-18:
thanks @pjreed for the clarification.

I opened https://github.com/ros-infrastructure/ros_buildfarm_config/pull/111 to disable the build on armhf. Feel free to create such PRs yourself in the future as this will be a more efficient process.

Cheers!

pjreed, 2018-04-18:
Ok, thanks!

closed 2018-04-18 by pjreed