swri-robotics/mapviz #581: nav_msgs/path not showing, no transform between  and map

alex94263, 2018-05-07:
I am trying to use MapViz to show a path published on a nav_msgs/Path topic.  
GPS location through a NavSatFix message is displayed fine.  
The part used to launch mapviz:  
```<node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /dummy 100"  />
    <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" >
        <param name="local_xy_frame" value="/map"/>
        <param name="local_xy_origin" value="Spata"/>
        <rosparam param="local_xy_origins">
            [{ name: Spata,
             latitude: 52.3079338074,
             longitude: 13.6578741074,
             altitude: 10,
             heading: 0.0}]
        </rosparam>
    </node>

    <node pkg="mapviz" type="mapviz" name="mapviz"/>
```
As i did not find any documentation or other posts explaining this issue, i would very much appreciate it if someone could explain how to implement the missing transformation or how to use the path topic differently.

malban, 2018-05-07:
The path message will need to have a frame id at the top level, in this case "/wgs84".  It probably should also have a valid timestamp.

alex94263, 2018-05-11:
got it running with 
```
header: 
  seq: 1
  stamp: 
    secs: 1526024893
    nsecs: 913650035
  frame_id: "/wgs84"
poses: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 1526024893
        nsecs: 913722038
      frame_id: "/wgs84"
    pose: 
      position: 
        x: 13.6584415021
        y: 52.3075536612
        z: 10.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
  - 
    header: 
      seq: 1
      stamp: 
        secs: 1526024893
        nsecs: 913795948
      frame_id: "/wgs84"
    pose: 
      position: 
        x: 13.6581506913
        y: 52.307588447
        z: 10.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
  - 
    header: 
      seq: 2
      stamp: 
        secs: 1526024893
        nsecs: 913839101
      frame_id: "/wgs84"
    pose: 
      position: 
        x: 13.6578741074
        y: 52.3079338074
        z: 10.0
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
---
```
thank you for clarification

closed 2018-05-11 by alex94263