swri-robotics/mapviz #584: [help_wanted] multi_res_image: image not displayed

alex94263, 2018-06-01:
Hello,
I wanted to show a top_down view of a simulation environment as "map", for demonstration purposes.
However I can't seem to get it to be displayed over the multi_res_image plugin.

[test1.geo.txt](https://github.com/swri-robotics/mapviz/files/2062517/test1.geo.txt)

the launch script used is the following
```
<launch>

     <env name="ROSCONSOLE_FORMAT" value="[${thread}] [${node}/${function}:${line}]: ${message}"/>
  <node pkg="mapviz" type="mapviz" name="$(anon mapviz)" output="screen"/>

  <node pkg="swri_transform_util" type="initialize_origin.py" name="initialize_origin" output="screen">
    <param name="local_xy_frame" value="/map"/>
    <param name="local_xy_origin" value="swri"/>
    <rosparam param="local_xy_origins">
      [{ name: swri,
         latitude: 48.8926980617,
         longitude: 9.17283197667,
         altitude: 306.471004486,
         heading: 0.0}]
    </rosparam>
  </node>

  <node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /utm /map 100" output="screen"  />
</launch>
```

Now what confuses me the most, is in a directory structure like this
|-test1.geo
|-tiles
   └---layer0
            └-----tile00000x00000.jpg

the launch works fine and produces status OK, but not showing the Image.
Removing the Image from the directory, renaming layer0, moving tile00000x00000.jpg to tiles-directory or setting the image path to "tiles/layer0" results in the plugin not being able to load the image.
``` Error: Failed to load image.```
Also an empty tiles directory produces the same error.

Is this unability to load the file due to the file being a "simple" jpg file?

Thank you in advance for your help.

malban, 2018-06-01:
It looks like there are two issues:

1. The local origin at (48.94 lat, 9.17 lon) appears to be pretty far off from the image geo reference (43 lat, 11.21 lon), if I am reading the UTM easting and northing values correctly.   I'm assuming the image is in the same UTM zone as the origin you defined.  Mapviz will be centered on the target frame by default.  So if you specify the target frame as 'map' if will be centered on the local origin and it may be very difficult to zoom/scroll to the correct location.  I would set the origin to be the center of the area you are mapping  (43 lat, 11.21 lon).

2. The static transform between /utm and /map is confusing things.  Mapviz is already capable of correctly converting between data specified in UTM coordinates and TF assuming the origin is set correctly and the data is given the '/utm' frame id.  The static transform defined here in effect says that at the 'map' origin (0,0) which is also (48.94 lat, 9.17 lon) the UTM coordinates are (0 easting, 0 northing) which is incorrect.  I would replace that static transform with this one, which is just an identity to make mapviz happy in the case no other TF data has been published:

```
<node pkg="tf" type="static_transform_publisher" name="swri_transform" args="0 0 0 0 0 0 /map /origin 100" output="screen"  />
```

After making those changes I was able to see a dummy image I created using your geo and launch files.

alex94263, 2018-06-01:
Thank you very much, yes the combination of both made the problem.

closed 2018-06-01 by alex94263