swri-robotics/mapviz #585: Tile map plugin not showing map

pierbiag, 2018-06-04:
Hello, every time I try to have the map shown, it starts throwing Network errors at me and it doesn't display the map:

[7f21621b5880] [/mapviz_ubuntu_6740_9181524986721827937/ImageCache::NetworkError:251]: NETWORK ERROR

The error is always 251.

I have working internet connection on the machine I'm using.

Any idea on how I could fix this?

Thanks,
Pier

malban, 2018-06-04:
Can you upload the configuration file you are using?

pierbiag, 2018-06-04:
I was about to reply to my own issue. Despite trying to debug the problem for a few days, only now I decided to copy and paste the config in #575 ... and it works ':-)

I still get the 251 errors, but many less and the map now appears.

KindDragon, 2019-02-25:
Fixed in 1.1.0

daniel-dsouza, 2019-09-16:
Looks like this is resolved.

closed 2019-09-16 by daniel-dsouza