swri-robotics/mapviz #589: Release into ROS Melodic

pjreed, 2018-07-11:
This is currently waiting on [rosbridge_suite](https://github.com/RobotWebTools/rosbridge_suite/) to be released (see https://github.com/RobotWebTools/rosbridge_suite/issues/333 ) but after that should be ready to go.

clalancette, 2018-07-31:
It looks like this was released into Melodic back on July 19th, which is great :+1: .  However, it looks like the version that was released doesn't compile because of the pluginlib backwards compatibility problem, i.e. http://build.ros.org/view/Mbin_ds_dS64/job/Mbin_ds_dS64__mapviz_plugins__debian_stretch_amd64__binary/14/consoleFull .  In the meantime, that particular problem was fixed in 3edb1cf5e71a4b841d0f5e278004b62130b84a72 .  I confirmed that it compiles locally now.  Could we have another release into Melodic to try and fix that?  Thanks!

pjreed, 2018-07-31:
Oops, I actually thought I had made another release after that.  I'll do that shortly.

pjreed, 2018-07-31:
Should be all done now.

closed 2018-07-31 by pjreed

clalancette, 2018-07-31:
Excellent, thank you!