swri-robotics/mapviz #591: Display ECEF Path

ClementLeBihan, 2018-08-22:
Hi,

I'm trying to show a vehicle path in mapviz, but it's hard for me to understand how it works. 

I converted GPS info to ENU with a referencial point, with geodetic2enu MATLAB, to be able to integrate it in a Kalman Filter.

Now I have a nav::Path expressed in my ENU frame, and I don't know what should I do to show my Path on the map ...

I'm newbie with mapviz, and I didn't saw any real tutorial. I'm sorry this is not an issue about mapviz !

Best,
Clément

ClementLeBihan, 2018-08-22:
Ok I just watched the parameteres for ENU are in the launch file !
It solve my problem :D

closed 2018-08-22 by ClementLeBihan

malban, 2018-08-22:
@ClementLeBihan I'm not sure if the projection used by geodetic2enu will be the same as what is used by mapviz, even if the reference point in the same.  Near the reference point it may not be a problem.  Otherwise there may be discrepancies between your path and other geo-referenced data, such as satellite imagery or GPS readings.

Mapviz can support visualizing data that is lat/lon along with other TF data where x:longitude, y:latitude and the "/wgs84" frame is used.

ClementLeBihan, 2018-08-23:
Ok I got it :)

Best,
Clément