swri-robotics/mapviz #592: Can't start mapviz (process died on start)

greenfield932, 2018-09-17:
Hi, the problem is mapviz is died on start without gui appear. My ROS is kinetic, ubuntu 16.04.
Is there any need for predefined config file for mapviz, does it make default on start? It seems mapviz can't load, but I'm not sure if this is the only reason. 
My command: roslaunch mapviz mapviz.launch

Output:
[7efce2043880] [/mapviz_greenfield_OMEN_11766_7860604551760320339/Mapviz::Open:535]: Failed to load file: /home/greenfield/.mapviz_config
[7efce2043880] [/mapviz_greenfield_OMEN_11766_7860604551760320339/MapvizApplication::notify:51]: Unhandled std::exception in Qt event loop: Duration is out of dual 32-bit range
================================================================================REQUIRED process [mapviz_greenfield_OMEN_11766_7860604551760320339-1] has died!
process has died [pid 11783, exit code -11, cmd /opt/ros/kinetic/lib/mapviz/mapviz __name:=mapviz_greenfield_OMEN_11766_7860604551760320339 __log:=/home/greenfield/.ros/log/bc3bb2ec-ba41-11e8-b89f-34f64b2bc7e4/mapviz_greenfield_OMEN_11766_7860604551760320339-1.log].
log file: /home/greenfield/.ros/log/bc3bb2ec-ba41-11e8-b89f-34f64b2bc7e4/mapviz_greenfield_OMEN_11766_7860604551760320339-1*.log
Initiating shutdown!

pjreed, 2018-09-17:
mapviz launches fine for me, with or without a config file.  Did you have to do anything special to reproduce it?

malban, 2018-09-17:
It looks like the problem went away for me when I did a full apt-get update and upgrade.  It looks like one or more libraries were out of sync on my system.

@greenfield932 can you check if your system is up to date and see if you still see the issue?

kriskozak, 2018-09-17:
I've seen the error "Duration is out of dual 32-bit range" a couple of times recently, including with packages built from source. I'm not sure what is specifically causing it, but I'll note that it persisted until the respective workspaces were cleaned and rebuilt.

pjreed, 2018-09-17:
The most recent version of ROS seems like it broke binary compatibility with some of its libraries, and I've seen that manifest as duration-related errors.  It's worth doing a `sudo apt-get update && sudo apt-get upgrade` and then cleaning your workspace to see if the issues go away.

greenfield932, 2018-09-17:
Hi, guys,
thank you very much for such a quick response. I can't execute apt-get upgrade due to some reasons, I have complex environment which most likely will be broken after this command. Build from source worked for me, mapviz started fine.

closed 2018-09-17 by greenfield932