swri-robotics/mapviz #600: Crash Due to Unhandled Event

cagataysari, 2018-12-20:
Hello I am not sure if it is directly related to #336 but I think that I am having a crash due to an unhandled event in MapvizApplication. I am running ROS Kinetic on Ubuntu 16.04

Here is the backtrack:

```
(gdb) backtrace
#0  0x00007ffff7646f77 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#1  0x00007fffd8e22610 in QXcbConnection::handleXcbEvent(xcb_generic_event_t*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#2  0x00007fffd8e2305d in QXcbConnection::processXcbEvents() () from /usr/lib/x86_64-linux-gnu/libQt5XcbQpa.so.5
#3  0x00007fffef590c01 in QObject::event(QEvent*) () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#4  0x00007ffff75ea05c in QApplicationPrivate::notify_helper(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#5  0x00007ffff75ef516 in QApplication::notify(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5
#6  0x00007ffff7b983af in mapviz::MapvizApplication::notify(QObject*, QEvent*) ()
   from /opt/ros/kinetic/lib/librqt_mapviz.so
#7  0x00007fffef56138b in QCoreApplication::notifyInternal(QObject*, QEvent*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#8  0x00007fffef563786 in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#9  0x00007fffef5b73c3 in ?? () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#10 0x00007fffeeab8197 in g_main_context_dispatch () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#11 0x00007fffeeab83f0 in ?? () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#12 0x00007fffeeab849c in g_main_context_iteration () from /lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007fffef5b77cf in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#14 0x00007fffef55eb4a in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) ()
   from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#15 0x00007fffef566bec in QCoreApplication::exec() () from /usr/lib/x86_64-linux-gnu/libQt5Core.so.5
#16 0x00000000004051e4 in main ()
(gdb) 
```

pjreed, 2018-12-20:
Hmm, unfortunately, that stack trace doesn't really give any indication what event caused it... Is this easily reproducible for you?  If so, what steps do you go through to cause it?  Also, could you upload a copy of your mapviz configuration?  (it should be at `$HOME/.mapviz_config`)

cagataysari, 2018-12-21:
yes it is reproducable at any times. I am going to upload .mapviz_config, I don't have my pc with me. I saw #1  0x00007fffd8e22610 in QXcbConnection::handleXcbEvent(xcb_generic_event_t*) () , I thought It might be related to #336