swri-robotics/mapviz #618: Tile map plugin - Network error 203

tynowell, 2019-03-12:
Hello, 

I've got a fresh Melodic installation on Ubuntu 18.04 and I've installed mapviz (version 1.1.0) and the plugins via apt. (I have also built from source with the same results.) 

I'm having problems with the tile_map plugin. Adding a tile map, Stamen(Terrain) loads the center of the visible portion of the the map correctly but when panning out of the loaded area I get **'NETWORK ERROR: 203'** error. Some new sections load very slowly but old sections will disappear before they are out of the field of view. The faster the panning the faster the errors come so I'm guessing it might be a request frequency limitation(?). It never loads the full screen, just the central section of the FOV. The status is listed as **'OK'**.

I've also tried Bing maps (terrain) but no map sections are loaded after the api key is set. The following is my mapviz config file.

```
capture_directory: ~
fixed_frame: /wgs84
target_frame: <none>
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1853
window_height: 1172
view_scale: 7.187735e-05
offset_x: -10.76604
offset_y: -59.67392
use_latest_transforms: true
background: "#888a85"
displays:
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: <my-bing-maps-api-key>
      source: Bing Maps (terrain)
  - type: mapviz_plugins/tile_map
    name: new display
    config:
      visible: true
      collapsed: false
      custom_sources:
        []
      bing_api_key: ""
      source: Stamen (terrain)
```

(I normal add only one tile map display at a time, this config is just to show all settings)

The effect is shown below.

[image]

Some sections of the map (as you can see at the end of the GIF) never load.

I am also having a issue displaying GPS messages but I will create a separate issue as I don't know if they are related.

pjreed, 2019-03-12:
I see that your Fixed Frame is set to `/wgs84`; this will cause a lot of things to not render quite right since that is not actually a ROS tf frame.  Try setting this to whatever you set as the `local_xy_frame` for the `initialize_xy_origin` node; on a [standard ROS platform](http://www.ros.org/reps/rep-0105.html), this will be the `map` frame.  Do you see the same behavior if that's set?

tynowell, 2019-03-12:
Setting `local_xy_frame` to `/map` instead of `/wgs84` worked. Thanks.

It also had an effect on a related issue, #619 so I'll add more information to that.

closed 2019-03-12 by tynowell