swri-robotics/mapviz #619: No GPS messages received - NavSatFix and GPSFix

tynowell, 2019-03-12:
Hello,

The same installation is experiencing issue [#618](https://github.com/swri-robotics/mapviz/issues/618).

Adding navsat and gps displays in mapviz from their respective topics have the status **'No messages received'**. 

I have a live GPS stream from the nmea_navsat_driver package (with BU-353S4 GPS device) being publishing sensor_msgs/NavSatFix to /fix and a fix_translator from gps_common publishes gps_common/GPSFix in /gps from the /fix topic. Sample messages are shown below.

/fix (NavSatFix)
```
header: 
  seq: 885
  stamp: 
    secs: 1552399290
    nsecs: 267234086
  frame_id: "/wgs84"
status: 
  status: 0
  service: 1
latitude: 59.6673283333
longitude: 10.77517
altitude: 162.2
position_covariance: [0.6400000000000001, 0.0, 0.0, 0.0, 0.6400000000000001, 0.0, 0.0, 0.0, 2.5600000000000005]
position_covariance_type: 1
```

/gps (GPSFix)
```
header: 
  seq: 212
  stamp: 
    secs: 1552399775
    nsecs: 265425920
  frame_id: "/wgs84"
status: 
  header: 
    seq: 0
    stamp: 
      secs: 0
      nsecs:         0
    frame_id: ''
  satellites_used: 0
  satellite_used_prn: []
  satellites_visible: 0
  satellite_visible_prn: []
  satellite_visible_z: []
  satellite_visible_azimuth: []
  satellite_visible_snr: []
  status: 0
  motion_source: 1
  orientation_source: 1
  position_source: 1
latitude: 59.66729
longitude: 10.7751033333
altitude: 161.5
track: 0.0
speed: 0.0
climb: 0.0
pitch: 0.0
roll: 0.0
dip: 0.0
time: 0.0
gdop: 0.0
pdop: 0.0
hdop: 0.0
vdop: 0.0
tdop: 0.0
err: 0.0
err_horz: 0.0
err_vert: 0.0
err_track: 0.0
err_speed: 0.0
err_climb: 0.0
err_time: 0.0
err_pitch: 0.0
err_roll: 0.0
err_dip: 0.0
position_covariance: [0.6400000000000001, 0.0, 0.0, 0.0, 0.6400000000000001, 0.0, 0.0, 0.0, 2.5600000000000005]
position_covariance_type: 1
```

These are being published at 1Hz and appear to be valid messages. My mapviz configuration is as follows.

```
capture_directory: ~
fixed_frame: /wgs84
target_frame: <none>
fix_orientation: false
rotate_90: false
enable_antialiasing: true
show_displays: true
show_status_bar: true
show_capture_tools: true
window_width: 1853
window_height: 1025
view_scale: 4.296071e-05
offset_x: -10.77589
offset_y: -59.66535
use_latest_transforms: true
background: "#888a85"
displays:
  - type: mapviz_plugins/navsat
    name: new display
    config:
      visible: true
      collapsed: false
      topic: /fix
      color: "#00ff00"
      draw_style: lines
      position_tolerance: 0
      buffer_size: 0
  - type: mapviz_plugins/gps
    name: new display
    config:
      visible: true
      collapsed: false
      topic: /gps
      color: "#00ff00"
      draw_style: lines
      position_tolerance: 0
      buffer_size: 0
      show_laps: false
      static_arrow_sizes: false
      arrow_size: 25
```

Apart from the other issue with mapviz mentioned at the start, this computer has a fresh installation of Ubuntu and ROS and hasn't had any other issues. The launch file used is as follows.

```
<launch>
  <node pkg="nmea_navsat_driver" type="nmea_serial_driver" name="gps_device">
    <param name="frame_id" value="wgs84"/>
    <param name="useRMC" value="false"/>
  </node>

  <node name="fix_translator" pkg="gps_common" type="fix_translator">
      <!-- Translate from NavSatFix to GPSFix //-->
      <remap from="/navsat_fix_in"  to="/fix"/>
      <remap from="/gps_fix_out"    to="/gps"/>
  </node>

  <node pkg="mapviz" type="mapviz" name="mapviz" output="screen"/>
</launch>
```

pjreed, 2019-03-12:
I think that similar to #618, these plugins will have issues rendering if your Fixed Frame is set to `/wgs84`; try a world-fixed ROS tf frame instead (probably `map`) and see if they render properly then.

If you still don't see it render anything after switching frames, could you record a bag file with a few messages and attach it here?

tynowell, 2019-03-12:
Setting `local_xy_frame` to `/map` instead of `/wgs84` as well as the frame_id of the navsatfix messages fixed this issue as well as #618.

closed 2019-03-12 by tynowell

pjreed, 2019-03-12:
Great, good to hear that fixed the issue.

For what it's worth, the `frame_id` in the actual messages shouldn't matter; anything that receives NavSatFix or GPSFix messages should assume that their coordinates are in WGS84.