swri-robotics/mapviz #620: Show custom marker on map

Adrianrr94, 2019-03-13:
Hi,

I would like to add a custom marker like the blue car in the Readme's mapviz example.
[image]

Actually,what I want is exactly the same as the image, a drone's model following the obtained trayectory

I suppose that the blue car is a custom marker, so, how do I design a new visualization_msgs::Marker?

Thanks

tynowell, 2019-03-13:
Hi Adrianrr94,

I think what you are looking for is the `robot_image` plugin. You can provide an image of a drone it'll follow the assigned frame.

Adrianrr94, 2019-03-13:
Thanks Tyno, 

My fault, I read the readme but not carefully.

Thanks for your quick response.

closed 2019-03-13 by Adrianrr94