swri-robotics/mapviz #637: How to load Bing offline map

liyanli2345, 2019-07-05:
Can I configure it directly in the  file, or need to modify the code?

pjreed, 2019-07-10:
If you want to be able to view map tiles while you're offline, you'll need to use a tool like [MapProxy](https://mapproxy.org/) to create an offline cache and run your own maps server, or use something like [mapCreator](https://github.com/sjev/mapCreator) to combine tiles into a single and then configure Mapviz to use that as its tile source.  There's an example of somebody else doing something similar in #575, although I would like to better document how to do this.

liyanli2345, 2019-07-12:
Thanks a lot for your response.I  have implemented it with MapViz-Tile-Map-Google-Maps-Satellite，which actually uses MapProxy.